microsoft/devicescript #31: new icon

pelikhan, 2022-12-05:
the dino is cool but...

mmoskal, 2022-12-05:
final:
[image]
wip:
[image]
[image]
[image]

mmoskal, 2022-12-05:
[image]
[src.zip](https://github.com/microsoft/devicescript/files/10159045/src.zip)

closed 2022-12-05 by mmoskal