microsoft/devicescript #81: rename devsdbg to devicescriptdebugger

pelikhan, 2023-01-31:
we use full "devicescript" in other service names

[image]

pelikhan, 2023-02-01:
> I think you should use full name not camel name in tree. I want to rename camel devicescriptManager to devsmgr. These super long names get into constants in C which then don't fit on screen.

Right…. But abbreviation are harder to understand. Maybe another flag to control C constant generation?

mmoskal, 2023-02-01:
Then don't use abbreviations in the tree - just use the full service name. The lengthy constants also applies to TS...

closed 2023-02-02 by pelikhan