microsoft/devicescript #101: useshell on windows doesn't work

tballmsft, 2023-02-04:
```
PS C:\github\devicescript\vscode\sampleprj> yarn devicescript devtools --vscode
yarn : File C:\Users\tball\AppData\Roaming\npm\yarn.ps1 cannot be loaded 
because running scripts is disabled on this system. For more information, see 
about_Execution_Policies at https:/go.microsoft.com/fwlink/?LinkID=135170.    
At line:1 char:1
+ yarn devicescript devtools --vscode
+ ~~~~
    + CategoryInfo          : SecurityError: (:) [], PSSecurityException  
    + FullyQualifiedErrorId : UnauthorizedAcyarn devicescript devtools --vscode\github\devicescript\vscode\sampleprj>
yarn : File C:\Users\tball\AppData\Roaming\npm\yarn.ps1 cannot be loaded 
because running scripts is disabled on this system. For more 
information, see about_Execution_Policies at 
https:/go.microsoft.com/fwlink/?LinkID=135170.
```

pelikhan, 2023-02-04:
Gotta love that powershell terminal.

Assigned #101<https://nam06.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fmicrosoft%2Fdevicescript%2Fissues%2F101&data=05%7C01%7Cjhalleux%40microsoft.com%7C14b7433b55db4c0ffac508db06e98e8d%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C638111372410958692%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=esw8odQ9W0TugLPO7oNqaO97zqUfxu3QDmmkUBJlylk%3D&reserved=0> to @pelikhan<https://nam06.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fpelikhan&data=05%7C01%7Cjhalleux%40microsoft.com%7C14b7433b55db4c0ffac508db06e98e8d%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C638111372411115655%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=Fy2uTnPscf%2BiJjgHsXy8ELITCKzbVlC8ItldcrqrAfo%3D&reserved=0>.

tballmsft, 2023-02-04:
is there some other way to move forward on windows (i mean setting things up so that the devicescript script is allowed, without having to give blanket permission for all local scripts)? There must be a way

pelikhan, 2023-02-04:
Try changing your default terminal to something else than powershel

pelikhan, 2023-02-04:
anything

closed 2023-02-09 by pelikhan