microsoft/devicescript #107: getting started example doesn't work

tballmsft, 2023-02-09:
1. installed node18, vscode and latest extension
2. create mydevice project per instructions and copied code from https://microsoft.github.io/devicescript/getting-started/cli
3. run and debug, selecting main.ts and UI18 simulator
4. click "open simultaors" button
5. add barometer simulator (manually)
6. press "continue" in debug toolbar
7. see the following:

[image]

pelikhan, 2023-02-09:
are you running any code?

tballmsft, 2023-02-09:
yes, see step 3 - i am running the code in main.ts copied from the getting started:

```
import * as ds from "@devicescript/core"

console.log("starting...")
const sensor = new ds.AirPressure()
const mouse = new ds.HidMouse()
// listen for pressure changes
sensor.pressure.onChange(10, () => {
    // read sensor reading
    const pressure = sensor.pressure.read()
    console.log(pressure)
    // user blows in straw
    if (pressure > 1400) {
        // click!
        console.log(`click!`)
        mouse.setButton(ds.HidMouseButton.Left, ds.HidMouseButtonEvent.Click)
        // debouncing
        ds.sleepMs(50)
    }
})

```

tballmsft, 2023-02-09:
when i run the code, it stops at first line, then I start simulator, then I continue and get the error shown above.

pelikhan, 2023-02-09:
added to sampleprj