microsoft/devicescript #127: RFC: board support packages

mmoskal, 2023-02-10:
Idea:
* define a built-in mapping in DS CLI with stuff like `"pico_w": "https://github.com/microsoft/jacdac-pico/blob/main/boards/rp2040w/pico_w.board.json"`
* additional entries could be added in `devsconfig.json`
* for all these JSON files we generate .d.ts in `.devicescript/...` for intellisense
* for every target board/configuration, the user defines a `.ts` file (or puts everything in `main.ts` if only one target)
* for example, in `target_pico_w.ts` (or whatever)
```ts
import device from "@devicescript/pico_w" // export default
device.buttonA.subscribe(...) // assuming pico_w had a built-in button
```

On top of that we would allow adding services, also in `target_pico_w.ts`:
```ts
ds.startServer({
   service: "button",
   pin: 12 // only constants supported
})
```

We would collect all `ds.startServer()` and generate user-settings JSON at build time.

This way there is no need to introduce additional target configuration JSON files - everything is handled from TypeScript using existing "choose main file" mechanism (otherwise we'll have to add "choose target" mechanism in extension). I think this is easier to explain to TS people.

Note that board definition is still handled by plain JSON, it's only the user-provided services that would be defined in TS.

pelikhan, 2023-02-10:
TS on user side sound great.

pelikhan, 2023-02-10:
Do I get a client back when I start a service?

const btw = ds.startService(...)

mmoskal, 2023-02-10:
I was a bit worried you would then somehow have to pass that service down to modules you import but not I think it’s not that much of a problem.

pelikhan, 2023-02-10:
something like?
```ts
export const buttonA = ds.startStarvice({ .... })
```
woudl allow to import those downstream. Is that what you think?

mmoskal, 2023-02-10:
Let’s say you have a thermostat for two different boards. You will have two entry points a.ts and b.ts but otherwise shared code common.ts dealing with button. Both an and b would import common not the other way around. 

You can either just define a button role in common code and hope it binds correctly or have a function in common that stores roles somewhere. I guess this is fine. 

Otherwise we could do magic and have @devicescript/board resolve to different module based on some settings. But I think it would be just a mess. Let people have whatever config they need in common.ts using normal TS constructs.

mmoskal, 2023-02-10:
Ended up doing separate starting functions, otherwise error messages for invalid JSON config are bad.

```ts
    import * as ds from "@devicescript/core"

    function startRotaryEncoder(cfg: RotaryEncoderConfig): ds.RotaryEncoder
    function startButton(cfg: ButtonConfig): ds.Button
    function startRelay(cfg: RelayConfig): ds.Relay
    function startPower(cfg: PowerConfig): ds.Power
    function startLightLevel(cfg: LightLevelConfig): ds.LightLevel
    function startReflectedLight(cfg: ReflectedLightConfig): ds.ReflectedLight
    function startWaterLevel(cfg: WaterLevelConfig): ds.WaterLevel
    function startSoundLevel(cfg: SoundLevelConfig): ds.SoundLevel
    function startSoilMoisture(cfg: SoilMoistureConfig): ds.SoilMoisture
    function startPotentiometer(cfg: PotentiometerConfig): ds.Potentiometer
```

pelikhan, 2023-02-10:
did you use another module though?

    import * as ds from ***@***.***/core"

mmoskal, 2023-02-10:
Yes:

```ts
import { startButton } from "@devicescript/srvcfg"

const btn = startButton({
    pin: 2,
})

btn.down.subscribe(() => {
    console.log("down!")
})

```

pelikhan, 2023-02-10:
I found that in jacdac-ts, it was easier to clearly name "clients" as clients and "servers" as servers. In this case, are you starting a server or a clent?

(can we avoid acronyms and short names in module names, it's hard to understand the meaning, maybe simply `@devicescript/servers`)

pelikhan, 2023-02-10:
nit picking

mmoskal, 2023-02-11:
It's actually both client and server.

pelikhan, 2023-02-11:
oh right.

pelikhan, 2023-02-11:
`srvcfg` -> `servers`?

pelikhan, 2023-03-06:
@mmoskal done, close?

closed 2023-03-06 by mmoskal