microsoft/devicescript #174: devicescript build does not build package.json entrypoint

pelikhan, 2023-03-05:
It should look at package.json to find the default file.

```
$ devicescript build
using devs: v1.10.1, runtime: v6.2.0, node: v18.14.2 from /workspaces/devicescript/vscode/sampleprj/node_modules/@devicescript/cli/built
src/main.ts does not exist
Done in 0.45s.
@pelikhan ➜ .../vscode/sampleprj/packages/cowsay (rxjs) $ 
```

pelikhan, 2023-03-05:
see rxjs branch, run `yarn build` in `observables.

mmoskal, 2023-03-06:
Building package entry point would result in an empty binary (none of the functions are used at the top-level), so it's likely not what we need. I could change it so it finds any entry point (in this case `src/main-test.ts`) in `src/main.ts` is not found, or you can rename `main-test.ts` to `main.ts`

pelikhan, 2023-03-06:
Right, using main.ts instead of main-test.ts would work.

pelikhan, 2023-03-06:
Opted for using main.ts and not main-test.ts

closed 2023-03-06 by pelikhan