microsoft/devicescript #203: crash in cloud suite

pelikhan, 2023-03-11:
```
VM> upload data
VM> upload data
VM> upload data
VM> upload data
VM> upload data
VMERR> Aborted(native code called abort())
VMERR> RuntimeError: Aborted(native code called abort())
VMERR> Aborted(native code called abort())
VMERR> /Users/peli/Documents/devicescript/cli/built/devicescript-cli.cjs:12378
VMERR> 
VMERR> RuntimeError: Aborted(native code called abort())
VMERR> 
VMERR> Node.js v18.12.1
client: disconnected (2 clients)
VMERR> Exit code: 7 
VMERR>     WSSK-H: send cmd=0x98
VMERR>     WSSK-H: send cmd=0x98
VMERR>     WSSK-H: send cmd=0x98
VMERR>     WSSK-H: streaming expired
VMERR>     WSSK-H: send cmd=0x92
VMERR>     WSSK-H: send cmd=0x98
VMERR>     WSSK-H: send cmd=0x98
VMERR>     WSSK-H: send cmd=0x98
VMERR>     WSSK-H: streaming expired
VMERR>     WSSK-H: send cmd=0x92
VMERR>     WSSK-H: send cmd=0x98
VMERR>     WSSK-H: send cmd=0x98
VMERR>     WSSK-H: send cmd=0x98
VMERR>     WSSK-H: streaming expired
VMERR>     WSSK-H: send cmd=0x92
VMERR>     WSSK-H: send cmd=0x98
VMERR>     WSSK-H: send cmd=0x98
VMERR>     WSSK-H: send cmd=0x98
VMERR>     WSSK-H: streaming expired
VMERR>     WSSK-H: send cmd=0x92
VMERR>     WSSK-H: send cmd=0x98
VMERR>     WSSK-H: send cmd=0x98
VMERR>     WSSK-H: send cmd=0x98
VMERR>     WSSK-H: streaming expired
VMERR>     WSSK-H: send cmd=0x98
VMERR>     WSSK-H: send cmd=0x98
VMERR>     WSSK-H: send cmd=0x98
VMERR>     WSSK-H: send cmd=0x92
VMERR>     WSSK-H: streaming expired
VMERR>     WSSK-H: send cmd=0x98
VMERR>     WSSK-H: send cmd=0x98
VMERR>     WSSK-H: send cmd=0x98
VMERR>     WSSK-H: send cmd=0x92
VMERR>     WSSK-H: streaming expired
VMERR>     WSSK-H: send cmd=0x98
VMERR>     WSSK-H: send cmd=0x98
VMERR>     WSSK-H: send cmd=0x98
VMERR>     WSSK-H: send cmd=0x92
VMERR>     WSSK-H: streaming expired
VMERR>     WSSK-H: send cmd=0x98
VMERR>     WSSK-H: send cmd=0x98
VMERR>     WSSK-H: send cmd=0x98
VMERR>     assertion 'devs_gc_obj_valid(ctx, ptr)' failed at devicescript/value.c:92 in devs_value_from_pointer
VMERR>     at abort (/Users/peli/Documents/devicescript/cli/built/devicescript-cli.cjs:12949:19)
VMERR>     at _abort (/Users/peli/Documents/devicescript/cli/built/devicescript-cli.cjs:13285:11)
VMERR>     at hw_panic (wasm://wasm/000c6696:wasm-function[29]:0xde9)
VMERR>     at jd_assert_fail (wasm://wasm/000c6696:wasm-function[617]:0x1ee6a)
VMERR>     at devs_value_from_pointer (wasm://wasm/000c6696:wasm-function[357]:0x12b31)
VMERR>     at devs_jd_pkt_capture (wasm://wasm/000c6696:wasm-function[238]:0x9c39)
VMERR>     at devs_jd_wake_role (wasm://wasm/000c6696:wasm-function[236]:0x9b49)
VMERR>     at devs_jd_process_pkt (wasm://wasm/000c6696:wasm-function[242]:0xa39c)
VMERR>     at devs_client_event_handler (wasm://wasm/000c6696:wasm-function[80]:0x1e82)
VMERR>     at devsmgr_client_ev (wasm://wasm/000c6696:wasm-function[114]:0x473c)
VMERR>     at jd_client_handle_packet (wasm://wasm/000c6696:wasm-function[530]:0x1ad3f)
VMERR>     at jd_services_handle_packet (wasm://wasm/000c6696:wasm-function[584]:0x1dfb0)
VMERR>     at jd_process_everything_core (wasm://wasm/000c6696:wasm-function[596]:0x1e675)
VMERR>     at jd_process_everything (wasm://wasm/000c6696:wasm-function[595]:0x1e62e)
VMERR>     at jd_em_process (wasm://wasm/000c6696:wasm-function[46]:0xf5b)
VMERR>     at Object._jd_em_process (/Users/peli/Documents/devicescript/cli/built/devicescript-cli.cjs:13008:31)
VMERR>     at Timeout._onTimeout (/Users/peli/Documents/devicescript/cli/built/devicescript-cli.cjs:12259:25)
VMERR>     at listOnTimeout (node:internal/timers:564:17)
VMERR>     at process.processTimers (node:internal/timers:507:7)
VMERR>     assertion '(ctx->flags & DEVS_CTX_FLAG_BUSY) == 0' failed at devicescript/devicescript.c:72 in devs_enter
VMERR>               throw ex;
VMERR>               ^
VMERR>     at abort (/Users/peli/Documents/devicescript/cli/built/devicescript-cli.cjs:12949:19)
VMERR>     at _abort (/Users/peli/Documents/devicescript/cli/built/devicescript-cli.cjs:13285:11)
VMERR>     at hw_panic (wasm://wasm/000c6696:wasm-function[29]:0xde9)
VMERR>     at jd_assert_fail (wasm://wasm/000c6696:wasm-function[617]:0x1ee6a)
VMERR>     at devs_client_event_handler (wasm://wasm/000c6696:wasm-function[80]:0x1f0c)
VMERR>     at devsmgr_client_ev (wasm://wasm/000c6696:wasm-function[114]:0x473c)
VMERR>     at jd_client_handle_packet (wasm://wasm/000c6696:wasm-function[530]:0x1ad3f)
VMERR>     at jd_services_handle_packet (wasm://wasm/000c6696:wasm-function[584]:0x1dfb0)
VMERR>     at jd_process_everything_core (wasm://wasm/000c6696:wasm-function[596]:0x1e675)
VMERR>     at jd_process_everything (wasm://wasm/000c6696:wasm-function[595]:0x1e62e)
VMERR>     at jd_em_process (wasm://wasm/000c6696:wasm-function[46]:0xf5b)
VMERR>     at Object._jd_em_process (/Users/peli/Documents/devicescript/cli/built/devicescript-cli.cjs:13008:31)
VMERR>     at Object.handlePacket (/Users/peli/Documents/devicescript/cli/built/devicescript-cli.cjs:12102:21)
VMERR>     at Socket.<anonymous> (/Users/peli/Documents/devicescript/cli/built/devicescript-cli.cjs:12160:29)
VMERR>     at Socket.emit (node:events:513:28)
VMERR>     at addChunk (node:internal/streams/readable:324:12)
VMERR>     at readableAddChunk (node:internal/streams/readable:297:9)
VMERR>     at Readable.push (node:internal/streams/readable:234:10)
VMERR>     at TCP.onStreamRead (node:internal/stream_base_commons:190:23)
VMERR> Exit code: 7 
```

mmoskal, 2023-03-13:
how do I repro this?

mmoskal, 2023-03-13:
likely fixed

commit 2fb72347 (2023-03-13): fix pointer checks; fixes #203
closed 2023-03-13 by mmoskal