microsoft/devicescript #205: Support for json in flash?

pelikhan, 2023-03-13:
Read only json loaded directly from flash?

mmoskal, 2023-03-16:
you can do ``` hex `00 01 ...` ```; what use case do you have in mind?

pelikhan, 2023-03-16:
Say i want to some readonly configuration data encoded as JSON. It would be great to have it stored in flash and not use head at runtime.

```
const configThatNeverChagnes = json`{
  "adsfasf": "sdfsdf"
}`
```

mmoskal, 2023-03-16:
I guess it would be more like

```ts
const cfg = Object.freeze({ "foo": 1 })
```

or maybe `Object.deepFreeze()`.

I don't think this is critical though...

pelikhan, 2023-03-16:
In MakeCode, we've had several instances where people had massive array of numbers to implement a driver. These arrays we using heap for now good reason. I would be good to be able to encode these kind of lookup tables in flash.... So maybe not object but something like `number[]`