microsoft/devicescript #210: onChange handler does not exist

tballmsft, 2023-03-13:
Following the guide and am getting this error from TSC:

[image]

tballmsft, 2023-03-13:
compiling also fails 
```
C:\Github\mydevice>devs devtools src/main.ts
using devs: v1.10.3, runtime: v6.3.0, node: v18.12.1 from C:\Github\mydevice\node_modules\@devicescript\cli\built
devs: v1.10.3, runtime: v6.3.0, node: v18.12.1
   http     : http://127.0.0.1:8081
   websocket: ws://127.0.0.1:8081
   tcpsocket: tcp://127.0.0.1:8082
   dbgserver: tcp://127.0.0.1:8083
building src/main.ts...
build src/main.ts Failed
src\main.ts:7:17 - error TS2339: Property 'onChange' does not exist on type 'Register<number>'.

7 sensor.pressure.onChange(10, async () => {
                  ~~~~~~~~
watching src/main.ts...
```

tballmsft, 2023-03-13:
this compiles. Should the example be changed?

```
sensor.pressure.subscribe(async () => {
    // read sensor reading
...
```

pelikhan, 2023-03-13:
https://github.com/microsoft/devicescript/commit/6a74959590ee4765727ac90be6697debe4de593d

closed 2023-03-13 by pelikhan