microsoft/devicescript #211: "Start Simulator" button does nothing

tballmsft, 2023-03-13:
I've got code compile and under watch, but "Start Simulator" button does nothing (even though it is enabled):

[image]

pelikhan, 2023-03-14:
[image]

pelikhan, 2023-03-14:
@mmoskal it seems that we are trying to be too smart about dependenceis on @devicescript/compiler (parseAnyInt leaked into vscode).

pelikhan, 2023-03-14:
fixed

closed 2023-03-14 by pelikhan

mmoskal, 2023-03-14:
we could just always embed `@devicescript/compiler` instead of loading it from node_modules

pelikhan, 2023-03-14:
The solution was to use @devicescript/interop