microsoft/devicescript #216: sensor.pressure.subscribe not working

tballmsft, 2023-03-14:
I've started up the sample program and got barometer pressure simulator up and running and bound, but the program is not responding to changes in pressure.

[image]

pelikhan, 2023-03-14:
It looks like the program is paused on the first line of the program.

Assigned #216<https://nam06.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fmicrosoft%2Fdevicescript%2Fissues%2F216&data=05%7C01%7Cjhalleux%40microsoft.com%7C8078150a93634517b45f08db24bb5786%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C638144159275100161%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=1C2uHlXxbYGQPk0Eo%2B79ZqHA4pldYvXNPwGCLH00mgo%3D&reserved=0> to @pelikhan<https://nam06.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fpelikhan&data=05%7C01%7Cjhalleux%40microsoft.com%7C8078150a93634517b45f08db24bb5786%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C638144159275100161%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=uQLxxsQd0sucrVXH7nkaKSrOy5f%2B%2Fk6zKoH6grMMz7k%3D&reserved=0>.

pelikhan, 2023-03-14:
Please attach your project source to the bug.

pelikhan, 2023-03-14:
we need a better example that does not use hidmouse,

tballmsft, 2023-03-14:
Ah, thanks for the find. When I run I get this error now:

[image]

tballmsft, 2023-03-14:
I am wondering if the underlying runtime is outdated:
```
VM> DeviceScript Simulator (WASM) v6.3.0; file v6.3.0
VM> starting...
VM> 2520.367
VM> click!
VM> 2520.367
VM> click!
VMERR> Aborted(native code called abort())
VMERR> C:\Github\mydevice\node_modules\@devicescript\cli\built\devicescript-cli.cjs:12382
VMERR>
VMERR> RuntimeError: Aborted(native code called abort())
VMERR>
VMERR> Node.js v18.12.1
Error: read ECONNRESET
    at TCP.onStreamRead (node:internal/stream_base_commons:217:20) {
  errno: -4077,
  code: 'ECONNRESET',
  syscall: 'read'
}
VMERR> Exit code: 7 
VMERR> using devs: v1.10.3, runtime: v6.3.0, node: v18.12.1 from C:\Github\mydevice\node_modules\@devicescript\cli\built
VMERR> connected to 127.0.0.1:8082
VMERR>     self-device: 441fdf979598f896/KK76
VMERR>     device created: KK76
VMERR>     device created: BG51
VMERR>     device created: TK06
VMERR>     device created: VC82
VMERR>     device created: UB15
VMERR>     set role sensor -> UB15:1
VMERR>     dbg: suspend 6
VMERR>     dbg: suspend 2
VMERR>     JD_PANIC() at devicescript/objects.c:68 in devs_maplike_iter
VMERR>               throw ex;
VMERR>               ^
VMERR>     at abort (C:\Github\mydevice\node_modules\@devicescript\cli\built\devicescript-cli.cjs:12953:19) 
VMERR>     at _abort (C:\Github\mydevice\node_modules\@devicescript\cli\built\devicescript-cli.cjs:13289:11)VMERR>     at hw_panic (wasm://wasm/000c5152:wasm-function[29]:0xe06)
VMERR>     at jd_panic_core (wasm://wasm/000c5152:wasm-function[609]:0x1e70d)
VMERR>     at devs_maplike_iter (wasm://wasm/000c5152:wasm-function[261]:0xc05c)
VMERR>     at obj_get_props (wasm://wasm/000c5152:wasm-function[94]:0x320d)
VMERR>     at expand_value (wasm://wasm/000c5152:wasm-function[95]:0x33b0)
VMERR>     at expand_key_value (wasm://wasm/000c5152:wasm-function[98]:0x3759)
VMERR>     at obj_get_props (wasm://wasm/000c5152:wasm-function[94]:0x31cc)
VMERR>     at devsdbg_handle_packet (wasm://wasm/000c5152:wasm-function[90]:0x27a8)
VMERR>     at jd_services_handle_packet (wasm://wasm/000c5152:wasm-function[581]:0x1dbde)
VMERR>     at jd_process_everything_core (wasm://wasm/000c5152:wasm-function[593]:0x1e1af)
VMERR>     at jd_process_everything (wasm://wasm/000c5152:wasm-function[592]:0x1e168)
VMERR>     at jd_em_process (wasm://wasm/000c5152:wasm-function[46]:0xf76)
VMERR>     at Object._jd_em_process (C:\Github\mydevice\node_modules\@devicescript\cli\built\devicescript-cli.cjs:13012:31)
VMERR>     at Object.handlePacket (C:\Github\mydevice\node_modules\@devicescript\cli\built\devicescript-cli.cjs:12102:21)
VMERR>     at Socket.<anonymous> (C:\Github\mydevice\node_modules\@devicescript\cli\built\devicescript-cli.cjs:12160:29)
VMERR>     at Socket.emit (node:events:513:28)
VMERR>     at addChunk (node:internal/streams/readable:324:12)
VMERR>     at readableAddChunk (node:internal/streams/readable:297:9)
VMERR>     at Readable.push (node:internal/streams/readable:234:10)
VMERR>     at TCP.onStreamRead (node:internal/stream_base_commons:190:23)
VMERR> Exit code: 7 
dbgserver: connection closed

````

pelikhan, 2023-03-14:
I think Michal needed to look at this. Look like it got tree-shakes.

VMERR> ***@***.***\cli\built\devicescript-cli.cjs:12382
VMERR> using devs: v1.10.3, runtime: v6.3.0, node: v18.12.1 from ***@***.***\cli\built
VMERR>     at abort ***@***.***\cli\built\devicescript-cli.cjs:12953:19)
VMERR>     at _abort ***@***.***\cli\built\devicescript-cli.cjs:13289:11)VMERR>     at hw_panic (wasm://wasm/000c5152:wasm-function[29]:0xe06)
VMERR>     at Object._jd_em_process ***@***.***\cli\built\devicescript-cli.cjs:13012:31)
VMERR>     at Object.handlePacket ***@***.***\cli\built\devicescript-cli.cjs:12102:21)
VMERR>     at Socket.<anonymous> ***@***.***\cli\built\devicescript-cli.cjs:12160:29)

tballmsft, 2023-03-14:
The project source is (more or less) the sample from the getting started guide:
```
import * as ds from "@devicescript/core"

console.log("starting...")
const sensor = new ds.AirPressure()
const mouse = new ds.HidMouse()
// listen for pressure changes
sensor.pressure.subscribe(async () => {
    // read sensor reading
    const pressure = await sensor.pressure.read()
    console.log(pressure)
    // user blows in straw
    if (pressure > 1400) {
        // click!
        console.log(`click!`)
        await mouse.setButton(
            ds.HidMouseButton.Left,
            ds.HidMouseButtonEvent.Click
        )
        // debouncing
        await ds.sleepMs(50)
    }
})

```

tballmsft, 2023-03-14:
started from empty project. works now

closed 2023-03-14 by tballmsft