microsoft/devicescript #220: something like this?

tballmsft, 2023-03-14:
Here's start of a very simple sample with input (sensor) and output (traffic light).

```
import * as ds from "@devicescript/core"

console.log("starting...")
const temp = new ds.Temperature()
const light = new ds.TrafficLight()
// listen for pressure changes
temp.temperature.subscribe(async t => {
    console.log(t)
    if (t > 30) {
        await light.green.write(false)
        await light.yellow.write(false)
        await light.red.write(true)
    } else {
        await light.green.write(true)
        await light.yellow.write(false)
        await light.red.write(false)
    }
})

```

tballmsft, 2023-03-14:
[image]

pelikhan, 2023-07-21:
added sample https://github.com/microsoft/devicescript/commit/2544de05e56455a0f7abd3252adb1a994a6d1920

closed 2023-07-21 by pelikhan