microsoft/devicescript #231: option to avoid stopping on entry with the debugger

pelikhan, 2023-03-16:
Start debugger without stopping on entry

mmoskal, 2023-03-17:
implemented in debugger - you can now set it in JSON, but needs UI

pelikhan, 2023-03-17:
Typically one needs to edit launch.json.

Assigned #231<https://nam06.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fmicrosoft%2Fdevicescript%2Fissues%2F231&data=05%7C01%7Cjhalleux%40microsoft.com%7C42ff1c1ad2834525e8b908db267e8f38%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C638146097235867380%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=AaRYo4mdaNFgOdQjirfvQ9r9dXB7LNdxjEvuNrmDJkQ%3D&reserved=0> to @pelikhan<https://nam06.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fpelikhan&data=05%7C01%7Cjhalleux%40microsoft.com%7C42ff1c1ad2834525e8b908db267e8f38%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C638146097235867380%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=3TY%2Fl41J3lG3IalJ%2FRnarUoEMMUzPyaSkHmk0WKof3Y%3D&reserved=0>.

closed 2023-03-17 by pelikhan