microsoft/devicescript #233: Version mismatch software upload

louisvangeldrop, 2023-03-17:
During deploying main.ts or Start Debugger I get the following error message:

DEV: ! version mismatch
DEV: * Raspberry Pi Pico W v6.0.0; file v6.5.0

Deploy cancelled. Your device firmware (v6.0.0) is outdated (min v6.5.0).

pelikhan, 2023-03-17:
Hi Louis,

Our veresion detection is working :) We need a newer firmware build for the pico. Do you have an ESP32? That one should be up to date.

@mmoskal do you have a recent pico build?

louisvangeldrop, 2023-03-18:
Hello Peli,
I have just flashed an ESP32-Wroom mcu and it works now. In the debugger the software doesn't stop at the breakpoints. However I can step through the code. Variables are not shown in the debug window.

I also have an ESP32C3 Lilygo TTGO and some ESP32S2 based mcu's. I will try later today.

pelikhan, 2023-03-18:
Thanks for reporting! (https://github.com/microsoft/devicescript/issues/219)

pelikhan, 2023-03-18:
We are tracking issues with the compiled tools on npm. You might want to try the developer setup for now.

louisvangeldrop, 2023-03-20:
I am using the VSCode extension and connect to an ESP32Wroom Wemos board. 
In the Terminal Output tab I can see that my program works - it connects to my Wifi router and returns the found IP-address. The I loop every second in order to log the IP-address
Starting the debugger stops at the first line of the program and I can step through the program using the F10-key. 
However it doesn't stop at the selected breakpoint when I press the F5 key.
In the Terminal tab I get the following error:
DEV> USB: second begin
DEV> USB: mismatched stop
DEV> USB: crc err

mmoskal, 2023-03-20:
This looks like a problem with the UART chip on the board. Do you know which UART chip does it use? I actually had good results with CP2102 and CH340.

Also, S2 or C3 (provided they have the USB connected directly) will work better.

I'll try to get the RP2040 building going again soon!

mmoskal, 2023-03-22:
@pelikhan yeah, once we know what they are; I don't think we have big enough sample size just yet

mmoskal, 2023-03-22:
the rp2040 should be now good to go

closed 2023-03-22 by mmoskal