microsoft/devicescript #273: string.toFixed

pelikhan, 2023-03-25:
tends to land in generated code.

```
   console.log(
        `Temperature: ${temperature.toFixed(2)}°C, Humidity: ${humidity.toFixed(
            2
        )}%`
    )
 
```

JackTam1993, 2023-09-14:
I added a function called toFixed and I wrote a test case to test it. when I run the test function, it returned message: `setting field 'toFixed' of a builtin frozen object`.

pelikhan, 2023-09-14:
Do you have a pull request we could look at?

JackTam1993, 2023-09-15:
https://github.com/microsoft/devicescript/pull/621 here is the pull request. Thank you.

pelikhan, 2023-09-15:
@mmoskal is number prototype special handled by the compiler?

mmoskal, 2023-11-01:
See comment on PR https://github.com/microsoft/devicescript/pull/621#issuecomment-1789310339