microsoft/devicescript #275: console.log of objects

pelikhan, 2023-03-25:
A cool logging trick in js is to structure variables into an object before logging. This prints 'map'.

```
    const pid = await readWord()
    console.log({ pid })
```

mmoskal, 2023-03-25:
Do we want `"" + { pid }` to return `"[Object]"` or `"{ pid: 12 }"`? (def. not current `"map"`)
I think the second option is better than special-casing `console.log()`.

pelikhan, 2023-03-25:
Imo we should err towards matching v8 behavior , so (1)

pelikhan, 2023-03-25:
But console.log({pid}) should expand it

mmoskal, 2023-03-25:
IMHO `.toString()` returning `"[Object]"` is one of the most useless "features" of JS. The only reason to support it would be  code like `if (x.toString()  == "[Object]")` in some legacy libraries, but we do not support legacy libraries anyways and such code is brittle beyond repair in V8 as well (eg. overriding `.toString()`).

pelikhan, 2023-03-25:
yes, but that's what seasoned web dev expect. Also a large object could accidentally be converted to json, not sure that's wanted.

mmoskal, 2023-03-25:
They expect it - so they won't even try to run `.toString()` on objects, so they won't be surprised.

It would have limits like `util.inspect()` in node.

Special-casing `console.log()` means ``` console.log(`foo: ${x}`) ``` is quite different from `console.log("foo:", x)`, and also you can't do `mylog("foo:" + x)`

mmoskal, 2023-03-26:
It currently applies to `console.log()` and template strings but not to "+" unless used in `console.log()` - I guess we need to clean this up later

commit e6e9187f (2023-03-26): implement object inspection; fixes #275
closed 2023-03-26 by mmoskal