microsoft/devicescript #306: m5stack atom not recognized

louisvangeldrop, 2023-03-30:
I have a m5stack atom matrix mcu with a FTDI serial driver. See https://docs.m5stack.com/en/core/atom_matrix
The m5stack is unfortunately not recognized as a webserial device in the browser.

louisvangeldrop, 2023-03-30:
Just tried m5atom in jacdac docs in developer mode and it is recognized now.
Is there an escape "developer mode" option available in devicescript?

mmoskal, 2023-03-30:
There is `--all-serial` for `devs flash esp32`; `Flags.developerMode` in `jacdac-ts` should enable it just like on the website but I don't think we expose it now

pelikhan, 2023-03-30:
I'm adding `--dev` to enable the developer mode generally.

pelikhan, 2023-03-31:
You will find a "developer mode" in the settings of the extension. It order for it to work, make sure to update the @devicescript/cli reference to the  latest build

```
yarn upgrade @devicescript/cli@latest
```

louisvangeldrop, 2023-03-31:
After installing 2.2.25, the command "npx --yes @devicescript/cli@latest init" failed. "Unable to find devicescript".

I installed version 2.2.24 and now the --dev option works.
 
However I still do get the JDUSB errors. 
JDUSB Error: second begin
JDUSB Error: dual magic
JDUSB Error: short frm
JDUSB Error: invalid quote
JDUSB Error: mismatched stop
JDUSB Error: invalid quote
JDUSB Error: mismatched stop

I am not sure if it has to do with the baud-rate. I also have problems with flashing at 15000000 baud. I reduced it to 115200.

pelikhan, 2023-03-31:
> After installing 2.2.25, the command "npx --yes @***@***.*** init" failed. "Unable to find devicescript".

There was a build error with 2.2.25 so that release is broken. Will investigate 2.2.26 which should have been fixed.

mmoskal, 2023-03-31:
I recommend not using external serial chips for now. There seem to be some issues with all of them, more with some. Can you use c3, s2 or rp2040?

pelikhan, 2023-03-31:
2.2.28 fixes the “devicescript” error - thabks for reporting!

louisvangeldrop, 2023-03-31:
I have noticed the problems with serial chips. I have a ESP32C3 LilyGo-T-OI-PLus board without CDC support.
This board always returns "no HF2 support"
My Lilygo ESP32_S2 board supports the USB- and the OTG-mode. It is working, but sometimes it is unclear with mode to select in order to get it working properly.