microsoft/devicescript #325: Error after deploying a new version of main.ts to rp2040_w

louisvangeldrop, 2023-04-03:
I am using a pico_w mcu. The following error occurs after deplying a new version to the pico_w device:

build src/main.ts OK
deploy status: OK
{
  context: 'serial',
  exception: JDError [JacdacError]: serial device 2E8A:420A (write error: Writing to COM port (GetOverlappedResult): Unknown error code 31)
      at NodeWebSerialIO.error (C:\Users\louis_000\OneDrive\Apps\repos\devicescript\wifi\node_modules\@devicescript\cli\built\devicescript-cli.cjs:50522:15)
      at C:\Users\louis_000\OneDrive\Apps\repos\devicescript\wifi\node_modules\@devicescript\cli\built\devicescript-cli.cjs:50539:14 
      at async C:\Users\louis_000\OneDrive\Apps\repos\devicescript\wifi\node_modules\@devicescript\cli\built\devicescript-cli.cjs:49955:9 {
    code: undefined,
    cancel: false
  }
}

pelikhan, 2023-04-03:
We had a little snafu of versions, can you try to flash again?

louisvangeldrop, 2023-04-03:
Just flashed with the latest version and it works now. THX

closed 2023-04-04 by pelikhan