microsoft/devicescript #344: pico_w crc error

louisvangeldrop, 2023-04-13:
- Connect to a pico_w using the site: http://localhost:8081
- Open the edge devtools
- In the console the message "cannot connect webclient ...." appears. This is correct, the development server is not running 
- This is the only message that is shown in the console
- Then I start VSCode with devicescript folder: ..../wifi project
- After a while, the connection to the development server is made. 
- The the following error is showing up in a loop:

(index):1  WebSocket connection to 'ws://localhost:8081/' failed: 
e @ (index):1
6812-bb808a282b52741081e3.js:9 ƒ (config) {
  var Module = config || {};

// The Module object: Our interface to the outside world. We import
// and export values on it. There are various ways Module can be used:
// 1. Not defined…
jdusb.ts:71  JDUSB Error: crc err
logError @ jdusb.ts:71
handleFrame @ jdusb.ts:89
decodeFrame @ jdusb.ts:146
JdUsbProto.io.onData @ jdusb.ts:66
(anonymous) @ webserialio.ts:133
asyncGeneratorStep @ asyncToGenerator.js:3
_next @ asyncToGenerator.js:25
Promise.then (async)
asyncGeneratorStep @ asyncToGenerator.js:13
_next @ asyncToGenerator.js:25
Promise.then (async)
asyncGeneratorStep @ asyncToGenerator.js:13
_next @ asyncToGenerator.js:25
Promise.then (async)
asyncGeneratorStep @ asyncToGenerator.js:13
_next @ asyncToGenerator.js:25
Promise.then (async)
asyncGeneratorStep @ asyncToGenerator.js:13
_next @ asyncToGenerator.js:25
(anonymous) @ asyncToGenerator.js:32
(anonymous) @ asyncToGenerator.js:21
readLoop @ webserialio.ts:9
(anonymous) @ webserialio.ts:231
asyncGeneratorStep @ asyncToGenerator.js:3
_next @ asyncToGenerator.js:25
Promise.then (async)
asyncGeneratorStep @ asyncToGenerator.js:13
_next @ asyncToGenerator.js:25
(anonymous) @ asyncToGenerator.js:32
(anonymous) @ asyncToGenerator.js:21
openDeviceAsync @ webserialio.ts:9
(anonymous) @ webserialio.ts:205
asyncGeneratorStep @ asyncToGenerator.js:3
_next @ asyncToGenerator.js:25
Promise.then (async)
asyncGeneratorStep @ asyncToGenerator.js:13
_next @ asyncToGenerator.js:25
(anonymous) @ asyncToGenerator.js:32
(anonymous) @ asyncToGenerator.js:21
connectAsync @ webserialio.ts:9
(anonymous) @ webserial.ts:49
asyncGeneratorStep @ asyncToGenerator.js:3
_next @ asyncToGenerator.js:25
(anonymous) @ asyncToGenerator.js:32
(anonymous) @ asyncToGenerator.js:21
transportConnectAsync @ webserial.ts:33
(anonymous) @ transport.ts:244
Promise.then (async)
connect @ transport.ts:243
(anonymous) @ transport.ts:328
asyncGeneratorStep @ asyncToGenerator.js:3
_next @ asyncToGenerator.js:25
Promise.then (async)
asyncGeneratorStep @ asyncToGenerator.js:13
_next @ asyncToGenerator.js:25
(anonymous) @ asyncToGenerator.js:32
(anonymous) @ asyncToGenerator.js:21
reconnect @ transport.ts:48
(anonymous) @ transport.ts:203
asyncGeneratorStep @ asyncToGenerator.js:3
_next @ asyncToGenerator.js:25
(anonymous) @ asyncToGenerator.js:32
(anonymous) @ asyncToGenerator.js:21
checkPulse @ transport.ts:48
emit @ eventsource.ts:258
(anonymous) @ bus.ts:646

pelikhan, 2023-04-14:
outdated firmware build?

louisvangeldrop, 2023-04-15:
I use this version: https://github.com/microsoft/devicescript-pico/releases/latest/download/devicescript-rp2040w-pico_w.uf2

pelikhan, 2023-04-25:
@louisvangeldrop is this still an issue for you?

louisvangeldrop, 2023-04-25:
I am on holidays and I don't have the pico-w with me.

pelikhan, 2023-07-22:
I'm going to close this one as I have been testing with a pico-w without issues for a while. Feel free to reopen.

closed 2023-07-22 by pelikhan