microsoft/devicescript #397: document memory/flash requirements

pelikhan, 2023-05-24:
Probably in the device section and intro.

aecorredor, 2023-05-24:
@pelikhan is Flash Encryption for ESP32 chips supported?: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/flash-encryption.html#flash-encryption-efuse.

If not, I'm wondering if we'd be able to activate Flash Encryption with ESP-IDF directly, and then keep developing with DeviceScript. I'm just unsure about `idf.py encrypted-flash` being supported in DeviceScript. It seems like you have your own flashing tooling.

pelikhan, 2023-05-24:
This one is for @mmoskal . For reference, the esp32 runtime is at https://github.com/microsoft/devicescript-esp32 and builds on top of ESP-IDF directly.

mmoskal, 2023-06-01:
The flash encryption Should-Just-Work(TM). We store user program in NVM but I think this is handled transparently. Having said that I have not tried...

commit bd279c7b (2023-06-22): add memory docs; fixes #397
closed 2023-06-22 by mmoskal