microsoft/devicescript #404: Add support for setStatusLight in pico-w

pelikhan, 2023-05-25:
Need to talk to the wireless chip to control that LED.

pelikhan, 2023-05-26:
The onboard LED does not seem to work on pico/pico-w

lukeh990, 2023-06-06:
I'm trying to use setStatusLight on a regular Pico and I get a error reading `JDUSB Error: crc err`

Here is what the code I'm running in `main.ts`:
```ts
import { setStatusLight } from "@devicescript/runtime";

console.log(":)");
await setStatusLight(1);
```

pelikhan, 2023-06-06:
Is this a fresh install? Could you try to flash the firmware again. We just did a release with sockets support.

import { setStatusLight } from ***@***.***/runtime";

pelikhan, 2023-06-06:
we might have been in the middle of upgrading the firware. could you try to upgrade the vscode extension and create a new project for the pico?

pelikhan, 2023-06-06:
@mmoskal it seems that setStatusLight is not working on the pico itself.

JackTam1993, 2023-06-08:
I met the same problem. the firmware and the vscode extension are the latest.

Raspberry Pi Pico W version: v2.10.900

closed 2023-06-23 by mmoskal