microsoft/devicescript #410: How to get on-board led pin information?

wh1tecat-nya, 2023-05-26:
I found a definition of led pin information in boards.json.
https://github.com/microsoft/devicescript/blob/9f8380259428f0f6290c9f7379bdfe3e06250418/compiler/src/boards.json#L695-L697

But it is not included `@dsboard/{boardname}(devicescript-boards.d.ts)`.
How to use on-board led or led pin assignment?

pedropalhari, 2023-05-26:
Hi, @pelikhan, thanks for the follow up!

Weirdly, I did encounter that section of the docs (and tried it). And on the RPi 2040 at least it's not (currently) working. It compiles and runs with no issue but nothing happens on the board. I tested on 2 different RPis I have here.

I was able to simulate a "hello world" with an external led though and didn't explore it much more (was just doing a quick getting started to check the framework) so I can't provide additional info now. But I can come back to it during the weekend and see if I'm able to pinpoint exactly what's happening.

pelikhan, 2023-05-26:
Fixed by https://github.com/microsoft/devicescript/commit/ecb7c7be5ab9c882e6194880e63d4ba60932fd06