microsoft/devicescript #414: Web server / client support?

zackify, 2023-05-26:
With the pico w, is there any chance of getting a simple web server like micro python has?

I know there is the local cloud stuff that is being worked on, but I would love being able to make a simple server or make http requests to things.

pelikhan, 2023-05-31:
Related to MQTT (https://github.com/microsoft/devicescript/issues/395), supporting different networking setups.

pelikhan, 2023-06-06:
We have added "fetch" and sockets in the latest release (v2.10.*)

zackify, 2023-06-06:
Wow that was fast! That’s so awesome thanks

zackify, 2023-06-06:
Any way for me to know when this will be in the docs? I didn’t see any notes on release either

pelikhan, 2023-06-06:
The docs are not quite there yet but the fetch implementation should give you a good idea of how to use sockets.

https://microsoft.github.io/devicescript/developer/net

pelikhan, 2023-06-06:
https://github.com/microsoft/devicescript/blob/main/packages/net/src/fetch.ts

zackify, 2023-06-06:
Thanks so much that’s perfect on the client side, just need to learn how I can start a http server next!