microsoft/devicescript #419: Esp32C3 Lilygo board not working

louisvangeldrop, 2023-05-30:
I have just flashed a Lilygo Esp32 C3 board using the esp32c3_bare board option adapted to the board settings according to the following info:
https://github.com/Xinyuan-LilyGO/LilyGo-T-OI-PLUS

Flashing went well, however when I Connect to Device using the Devicescript extension I receive the following error:
{
  transport: 'serial',
  context: 'connect',
  exception: JDError [JacdacError]: JDUSB: can't connect, no HF2 nor JDUSB; port=COM8, vendor: 1A86, product: 55D4
      at throwError (C:\Users\louis\OneDrive\Apps\Repos\IoT\devicescript\wifi\node_modules\@devicescript\cli\built\devicescript-cli.cjs:40566:13)
      at C:\Users\louis\OneDrive\Apps\Repos\IoT\devicescript\wifi\node_modules\@devicescript\cli\built\devicescript-cli.cjs:56389:7 {       
    code: 'transport/hf2-not-supported',
    cancel: false
  }
}
Any help is welcome, since this board has a built-in battery adapter, that I want to use for a DHT22 Temperature/Humidity sensor.

The board json-file:
{
    "$schema": "../esp32deviceconfig.schema.json",
    "devName": "lilygo",
    "productId": "0x3f140dcc",
    "$description": "A Battery 16340 holding ESP32C3 mcu .",
    "archId": "esp32c3",
    "url": "https://github.com/Xinyuan-LilyGO/LilyGo-T-OI-PLUS",
    "uartLog": true,
    "led": {
        "pin": 3
    },
    "pins": {
        "#pid": "0x3a1d89be",
        "A0": 2,
        "P4": 4,
        "P5": 5,
        "SCK": 6,
        "P7": 7,
        "P9": 9,
        "P10": 10,
        "RX": 20,
        "TX": 21
    },
    "services": [
        {
            "name": "buttonBOOT",
            "pin": 8,
            "service": "button"
        }
    ]
}

pelikhan, 2023-06-01:
Unfortunately, for the esp32-c3, we do not support external serial chips -- which is what is causing the problem here most likely.

closed 2023-06-01 by pelikhan

louisvangeldrop, 2023-06-01:
If I understand it well, it means that if I look in the device manager only USB Serial Devices are supported.