microsoft/devicescript #432: "loopback rx ovf" log spamming

Sharlottes, 2023-06-03:
**Describe the bug**
> A clear and concise description of what the bug is.

when I start DeviceScript with the code
```ts
import * as ds from "@devicescript/core"
import "@devicescript/observables"
import { startLightBulb, startServo } from "@devicescript/servers"
import { pins } from "@dsboard/adafruit_qt_py_c3"

const lightBulb = startLightBulb({
    pin: pins.A1,
})

const servo = startServo({
    pin: pins.A2,
})

setInterval(async () => {
    await lightBulb.toggle()
}, 500)

let i = 0
setInterval(async () => {
    i++
    await servo.angle.write((i % 180) - 90)
}, 1)
```

it spams "loopback rx ovf" text on the devicescript terminal and it's suddenly stoped and shows 3 error logs.
```txt
// ...and more "loopback rx ovf" log
DEV> loopback rx ovf
DEV> loopback rx ovf
DEV> loopback rx ovf
JDUSB Error: mismatched stop
DEV> loopback rx ovf
DEV> loopback rx ovf
DEV> loopback rx ovf
DEV> loopback rx ovf
DEV> loopback rx ovf
DEV> loopback rx ovf
DEV> loopback rx ovf
DEV> loopback rx ovf
DEV> loopback rx ovf
DEV> loopback rx ovf
DEV> loopback rx ovf
DEV> loopback rx ovf
DEV> loopback rx ovf
DEV> loopback rx ovf
DEV> loopback rx ovf
{
  context: 'serial',
  exception: JDError [JacdacError]: serial device 303A:1001 (write error: Writing to COM port (GetOverlappedResult): Unknown error code 31)
      at NodeWebSerialIO.error (C:\Users\user\Documents\GitHub\DSProject\node_modules\@devicescript\cli\built\devicescript-cli.cjs:57068:15)
      at C:\Users\user\Documents\GitHub\DSProject\node_modules\@devicescript\cli\built\devicescript-cli.cjs:57085:14  
      at async C:\Users\user\Documents\GitHub\DSProject\node_modules\@devicescript\cli\built\devicescript-cli.cjs:56501:9 {
    code: undefined,
    cancel: false
  }
}
{
  transport: 'serial',
  context: 'serial',
  exception: JDError [JacdacError]: serial device 303A:1001 (write error: Writing to COM port (GetOverlappedResult): Unknown error code 31)
      at NodeWebSerialIO.error (C:\Users\user\Documents\GitHub\DSProject\node_modules\@devicescript\cli\built\devicescript-cli.cjs:57068:15)
      at C:\Users\user\Documents\GitHub\DSProject\node_modules\@devicescript\cli\built\devicescript-cli.cjs:57085:14  
      at async C:\Users\user\Documents\GitHub\DSProject\node_modules\@devicescript\cli\built\devicescript-cli.cjs:56501:9 {
    code: undefined,
    cancel: false
  }
}
{
  context: 'serial',
  exception: JDError [JacdacError]: serial device n/a (Error: Writing to COM port (GetOverlappedResult): Unknown error code 31)
      at NodeWebSerialIO.error (C:\Users\user\Documents\GitHub\DSProject\node_modules\@devicescript\cli\built\devicescript-cli.cjs:57068:15)
      at SerialPort.<anonymous> (C:\Users\user\Documents\GitHub\DSProject\node_modules\@devicescript\cli\built\devicescript-cli.cjs:57124:16)
      at SerialPort.emit (node:events:513:28)
      at emitErrorNT (node:internal/streams/destroy:151:8) 
      at emitErrorCloseNT (node:internal/streams/destroy:116:3)
      at process.processTicksAndRejections (node:internal/process/task_queues:82:21) {
    code: undefined,
    cancel: false
  }
}
{
  transport: 'serial',
  context: 'serial',
  exception: JDError [JacdacError]: serial device n/a (Error: Writing to COM port (GetOverlappedResult): Unknown error code 31)
      at NodeWebSerialIO.error (C:\Users\user\Documents\GitHub\DSProject\node_modules\@devicescript\cli\built\devicescript-cli.cjs:57068:15)
      at SerialPort.<anonymous> (C:\Users\user\Documents\GitHub\DSProject\node_modules\@devicescript\cli\built\devicescript-cli.cjs:57124:16)
      at SerialPort.emit (node:events:513:28)
      at emitErrorNT (node:internal/streams/destroy:151:8) 
      at emitErrorCloseNT (node:internal/streams/destroy:116:3)
      at process.processTicksAndRejections (node:internal/process/task_queues:82:21) {
    code: undefined,
    cancel: false
  }
}
```
I reset the device (adafruit_qt_py_c3 board has reset button) and reconnect it by vscode extension.
[image]
then below logs are shown on devicescript terminal.
```txt
stopping serial transport
adding serial transport (requires "serialport" package)
```
after retrying several times(3), the terminal is suddenly died after showing the log at below image.
[image]
[image]

so I reconnect the device again by same behaviour.
and below logs are shown on the new devicescript terminal.
```txt
DeviceScript Development Server

using devs: v2.9.15, runtime: v2.9.15, node: v18.13.0 from C:\Users\user\Docu\Documents\GitHub\DSProject\node_modules\@devicescript\cli\built        
devs: v2.9.15, runtime: v2.9.15, node: v18.13.0 running in C:\Users\user\Docu\Documents\GitHub\DSProject
services: error adding accelerometer, classIdentifier already in use      
services: error adding brailledisplay, classIdentifier already in use     
services: error adding devicescriptmanager, classIdentifier already in usese
services: error adding distance, classIdentifier already in use
services: error adding ledstrip, classIdentifier already in use
services: error adding servo, classIdentifier already in use              
services: error adding temperature, classIdentifier already in use        
services: error adding vibrationmotor, classIdentifier already in use   
   http     : http://127.0.0.1:8081
   websocket: ws://127.0.0.1:8081
   tcpsocket: tcp://127.0.0.1:8082
   dbgserver: tcp://127.0.0.1:8083
webclient: connected (ws1, 1 clients)
webclient: connected (ws2, 2 clients)
adding serial transport (requires "serialport" package)
```

I realized that the devicescript code does not care about this issue because even if I reset the device and even if I re-flash the device, the same situation has occurred. i really don't know the reason of that "loopback rx ovf"

**To Reproduce**
> Steps to reproduce the behavior:
run that code in adafruit_qt_py_c3 device and click run button in vscode extension
[image]

**Expected behavior**
> A clear and concise description of what you expected to happen.
The LightBulb connected to the A1 pin should blink every 500 ms, and the servo motor connected to the A2 pin should rotate 1 degree every 1 ms.

**Screenshots**
> If applicable, add screenshots to help explain your problem.

**Additional context**
> Add any other context about the problem here.

I started devicescript a few days ago, and I'm new in the IoT space. so I may miss something important, please tell me if there is.
develop OS: windows10
develop environment: vscode, with devicescript extension and cli
develop device: [adafruit-qt-py-c3](https://microsoft.github.io/devicescript/devices/esp32/adafruit-qt-py-c3)
this issue occurred yesterday. I tried restarting everything. reset/reboot the device(board), restarting VScode, restarting the PC, recreating the project with editing nothing.
[image]

pelikhan, 2023-06-03:
Great bug report!

From a quick glance it seems that the second interval has a delay of 1ms. This is probably the issue (we should be robust to this).

Try increasing 1 to something like 50.

pelikhan, 2023-06-03:
@mmoskal can we update the error message with some kind of code that the vscode addon would expand into a help page? Seems like this one is a common gotcha.

Sharlottes, 2023-06-03:
> Great bug report!
> 
> From a quick glance it seems that the second interval has a delay of 1ms. This is probably the issue (we should be robust to this).
> 
> Try increasing 1 to something like 50.

I tried to increase 1 to 100 on the second try. but as I mentioned above, even if I created a new project(default generated code) same error(loopback rx ovf) occurred. it means the code is not related to this issue.
I also doubt if this is really deviceScript's problem, Because it happened all of a sudden without any clues. But I've never touched the board directly, and it's a new product that came two days ago, so it's not an aging issue.
Doubting hardware problems is the last thing to consider, so I wanted to ask for help with the GitHub issue.

Sharlottes, 2023-06-03:
if it was really code's problem, the issue should be fixed when I reset the board and start a plain `console.log(1);` code.

mmoskal, 2023-06-05:
@pelikhan as for the "loopback rx ovf" message, here is where it could be patched on CLI side https://github.com/microsoft/devicescript/blob/main/interop/src/debug.ts#L54

mmoskal, 2023-06-05:
we need to:
* [ ] wait say 3 seconds after restart before running user program; possibly wait more if the restart was due to abort()
* [ ] make devs cli more aggressive at connecting to the device (right now it just gives up at some point)
* [ ] throttle sending packets by the user
* [ ] provide a CLI command to clear the user program (and config unfortunately) using esptool

mmoskal, 2023-06-05:
@Sharlottes you should be able to erase the flash on the board by running `esptool erase_flash`. I have added `--clean` option to `devs flash` but it will take a while to get to you

pelikhan, 2023-06-06:
v2.10.2 has the --clean support. @Sharlottes

mmoskal, 2023-06-06:
the packets are now throttled in v2.10.500

Sharlottes, 2023-06-07:
thanks for your advice! I tried `devs flash --clean` and succeeded to clean and flash again.

<details>
<summary>show logs</summary>

```txt
user@DESKTOP-18FQSK1 MINGW64 ~/Documents/GitHub/DSProject (master)
$ devs flash --clean --board adafruit_qt_py_c3
using devs: v2.10.7, runtime: v2.10.7, node: v18.13.0 from C:\Users\user\AppData\Roaming\npm\node_modules\@devicescript\cli\built
using serial port COM3 at 1500000
run: py -m esptool
esptool: py -m esptool
fetch https://github.com/microsoft/devicescript-esp32/releases/latest/download/devicescript-esp32c3-adafruit_qt_py_c3-0x0.bin
saved .devicescript/cache/devicescript-esp32c3-adafruit_qt_py_c3-0x0.bin 1169856 bytes
run: py -m esptool --port COM3 --baud 1500000 erase_flash
esptool.py v4.6.1
Serial port COM3 
Connecting...    
Detecting chip type... ESP32-C3
Chip is ESP32-C3 (revision v0.4)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 34:85:18:17:3d:88
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 1500000
Changed.
Erasing flash (this may take a while)...
Chip erase completed successfully in 21.3s
Hard resetting via RTS pin...
erase flash OK!

user@DESKTOP-18FQSK1 MINGW64 ~/Documents/GitHub/DSProject (master)     
$ devs flash --board adafruit_qt_py_c3
using devs: v2.10.7, runtime: v2.10.7, node: v18.13.0 from C:\Users\user\AppData\Roaming\npm\node_modules\@devicescript\cli\built
using serial port COM3 at 1500000
run: py -m esptool
esptool: py -m esptool
using cached .devicescript/cache/devicescript-esp32c3-adafruit_qt_py_c3-0x0.bin
run: py -m esptool --port COM3 --baud 1500000 write_flash 0x0 .devicescript/cache/devicescript-esp32c3-adafruit_qt_py_c3-0x0.bin
esptool.py v4.6.1
Serial port COM3
Connecting...
Detecting chip type... ESP32-C3
Chip is ESP32-C3 (revision v0.4)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 34:85:18:17:3d:88
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 1500000
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x0011dfff...
Compressed 1169856 bytes to 651916...
Writing at 0x00000000... (2 %)
Writing at 0x0001af67... (7 %)
Writing at 0x0001fd7c... (10 %)
Writing at 0x000250d9... (12 %)
Writing at 0x0002e3db... (15 %)
Writing at 0x00038aec... (17 %)
Writing at 0x000419b0... (20 %)
Writing at 0x0004a1b6... (22 %)
Writing at 0x0005077e... (25 %)
Writing at 0x00057ae5... (27 %)
Writing at 0x0005f9ca... (30 %)
Writing at 0x000670ae... (32 %)
Writing at 0x0006dcae... (35 %)
Writing at 0x000749a4... (37 %)
Writing at 0x0007ba0e... (40 %)
Writing at 0x00082703... (42 %)
Writing at 0x00089748... (45 %)
Writing at 0x0008ffc4... (47 %)
Writing at 0x00096630... (50 %)
Writing at 0x0009ccce... (52 %)
Writing at 0x000a3bac... (55 %)
Writing at 0x000a9ff6... (57 %)
Writing at 0x000b0128... (60 %)
Writing at 0x000b5fc7... (62 %)
Writing at 0x000bc032... (65 %)
Writing at 0x000c26d6... (67 %)
Writing at 0x000c8cc2... (70 %)
Writing at 0x000cf090... (72 %)
Writing at 0x000d544d... (75 %)
Writing at 0x000db876... (77 %)
Writing at 0x000e1a40... (80 %)
Writing at 0x000e8132... (82 %)
Writing at 0x000ef0a6... (85 %)
Writing at 0x000f5f84... (87 %)
Writing at 0x000fc976... (90 %)
Writing at 0x00102d4a... (92 %)
Writing at 0x0010a76a... (95 %)
Writing at 0x0011151b... (97 %)
Writing at 0x001182ac... (100 %)
Wrote 1169856 bytes (651916 compressed) at 0x00000000 in 10.0 seco00 in 10.0 seconds (effective 934.9 kbit/s)...                    nds (
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
flash OK!
```
</details>

but after updating the DeviceScript, it suddenly required me to install node.js packages even if I already installed them again.
[image]

when I tried to connect the device with a new project, it still shows that message.
[image]
not only the "connect to device" button, but also the "start simulator" button, and plug/graph/dashboard/list icon are the same. 
[image]

also i tried to remove and install deviceScript extension again

pelikhan, 2023-06-07:
We moved our tooling to use `npm` instead of `yarn`. And I wonder if it is getting confused since you already have a `yarn.lock`.

VSCode is search for ./node_modules/.bin/devicescript . Could you should list of files in `node_modules/.bin` in that folder?

Sharlottes, 2023-06-07:
I tried to start the DeviceScript by CLI, not vscode extension because I want to see if the "loopback rx ovf" error is solved.
and..it works! 🎉 
[image]

> We moved our tooling to use npm instead of yarn. And I wonder if it is getting confused since you already have a yarn.lock.
I am also confused because [the document](https://microsoft.github.io/devicescript/getting-started/cli#launch-the-build-watch) mentioned yarn too. is it outdated?
[image]

there is no `.bin`.
[image]

after installing `npm i @devicescript/cli`, everything works fine. thanks for your advice again!

pelikhan, 2023-06-07:
yes the instructions were generated in a previous build which used yarn. We ran a workshop yesterday and a large amount of users had trouble with finding yarn on their machine; so we are moving away from it.

To be clear, did you install the CLI globally?

Sharlottes, 2023-06-07:
yes i did. should i remove global deviecscript CLI?

pelikhan, 2023-06-07:
Yes it would be best to remove it.

However, we should handle this case better…

Sharlottes, 2023-06-07:
```bash
user@DESKTOP-18FQSK1 MINGW64 ~/Documents/GitHub/DSProject (master)
$ npm uninstall -g @devicescript/cli

removed 128 packages, and audited 1 package in 578ms

found 0 vulnerabilities

user@DESKTOP-18FQSK1 MINGW64 ~/Documents/GitHub/DSProject (master)
$ devs
bash: devs: command not found
```

ok I uninstall cli which is installed globally. as I expected, it doesn't work even if there are "devs"-related files in `/node_modules/.bin/`.

<details>
<summary>node_modules/.bin/devs</summary>

```sh
#!/bin/sh
basedir=$(dirname "$(echo "$0" | sed -e 's,\\,/,g')")

case `uname` in
    *CYGWIN*|*MINGW*|*MSYS*) basedir=`cygpath -w "$basedir"`;;
esac

if [ -x "$basedir/node" ]; then
  exec "$basedir/node"  "$basedir/../@devicescript/cli/devicescript" "$@"
else 
  exec node  "$basedir/../@devicescript/cli/devicescript" "$@"
fi
```

</details>

<details>
<summary>node_modules/.bin/devs.cmd</summary>

```cmd
@ECHO off
GOTO start
:find_dp0
SET dp0=%~dp0
EXIT /b
:start
SETLOCAL
CALL :find_dp0

IF EXIST "%dp0%\node.exe" (
  SET "_prog=%dp0%\node.exe"
) ELSE (
  SET "_prog=node"
  SET PATHEXT=%PATHEXT:;.JS;=;%
)

endLocal & goto #_undefined_# 2>NUL || title %COMSPEC% & "%_prog%"  "%dp0%\..\@devicescript\cli\devicescript" %*
```

</details>

<details>
<summary>node_modules/.bin/devs.ps1</summary>

```ps1
#!/usr/bin/env pwsh
$basedir=Split-Path $MyInvocation.MyCommand.Definition -Parent

$exe=""
if ($PSVersionTable.PSVersion -lt "6.0" -or $IsWindows) {
  # Fix case when both the Windows and Linux builds of Node
  # are installed in the same directory
  $exe=".exe"
}
$ret=0
if (Test-Path "$basedir/node$exe") {
  # Support pipeline input
  if ($MyInvocation.ExpectingInput) {
    $input | & "$basedir/node$exe"  "$basedir/../@devicescript/cli/devicescript" $args
  } else {
    & "$basedir/node$exe"  "$basedir/../@devicescript/cli/devicescript" $args
  }
  $ret=$LASTEXITCODE
} else {
  # Support pipeline input
  if ($MyInvocation.ExpectingInput) {
    $input | & "node$exe"  "$basedir/../@devicescript/cli/devicescript" $args
  } else {
    & "node$exe"  "$basedir/../@devicescript/cli/devicescript" $args
  }
  $ret=$LASTEXITCODE
}
exit $ret
```

</details>

pelikhan, 2023-06-07:
finding the locally installed tools is somewhat flaky.

Try running 'npm run devs' to make it more robust. npm will find the devs tool path for you.

pelikhan, 2023-06-07:
Could you try to run install again? Just to be sure.

    npm install
    ls node_modules/.bin
    npm run devs

pelikhan, 2023-06-07:
Thanks for hanging in there!

pelikhan, 2023-06-07:
One more to see if yarn is smarter

    yarn devs

Sharlottes, 2023-06-07:
...oh my god. yarn is really smart.

```bash
user@DESKTOP-18FQSK1 MINGW64 ~/Documents/GitHub/DSProject (master)
$ yarn devs
yarn run v1.22.19
warning ..\..\..\package.json: No license field
$ C:\Users\user\Documents\GitHub\DSProject\node_modules\.bin\devs
using devs: v2.10.7, runtime: v2.10.7, node: v18.13.0 from C:\Users\user\Documents\GitHub\DSProject\node_modules\@devicescript\cli\built
Done in 1.94s.

user@DESKTOP-18FQSK1 MINGW64 ~/Documents/GitHub/DSProject (master)
$ devs
bash: devs: command not found

user@DESKTOP-18FQSK1 MINGW64 ~/Documents/GitHub/DSProject (master)
$ yarn devs --help
yarn run v1.22.19
warning ..\..\..\package.json: No license field
$ C:\Users\user\Documents\GitHub\DSProject\node_modules\.bin\devs --help
Usage: DeviceScript [options] [command]

build and run DeviceScript program https://aka.ms/devicescript

Options:
  -V, --version                                       output the version number
  -v, --verbose                                       more logging (can be repeated)
  --quiet                                             less logging
  --no-colors                                         disable color output
  --dev                                               developer mode
  -h, --help                                          display help for command

Commands:
  build [options] [src/mainXYZ.ts]                    build a DeviceScript file
  flags                                               show description of compiler flags
  devtools [options] [file.ts]                        launches a local development tools server
  run [options] [file.ts|file.devs]                   run a script
  vm [options]                                        start DeviceScript VM interpreter process
  disasm [options] [file.ts|file-dbg.json|file.devs]  disassemble .devs binary
  annotate                                            annotate stack frames in stdin
  flash [options]                                     flash DeviceScript runtime (interpreter/VM)
  add                                                 add a feature to the project
  init [options] [dir]                                creates or configures a devicescript project
  help [command]                                      display help for command
Done in 0.83s.
```

bash cannot find "devs" command itself but yarn can find the command. i might solve the problem roughly

pelikhan, 2023-06-07:
Fixed original issue. Let's use a new discussion for new problems.

closed 2023-06-07 by pelikhan