microsoft/devicescript #434: Raspberry Pi Pico can no longer be detected since today

TAMHAN, 2023-06-04:
**Describe the bug**
Yesterday,  VS Code was still able to detect my RP2040 based Raspberry Pi Pico on Ubuntu 20.04LTS. Today, it can no more. I changed nothing in the configuration.

**To Reproduce**
a) install firmware using the Visual Studio Code. This works well:

```
tamhan@TAMHAN18:~/devicescript/p1/DeviceScriptP1$ node ./node_modules/.bin/devicescript flash --board pico
using devs: v2.9.16, runtime: v2.9.16, node: v18.16.0 from /home/tamhan/devicescript/p1/DeviceScriptP1/node_modules/@devicescript/cli/built
no drives found; please plug your board while holding BOOTSEL button (or specify --drive)
Re-scanning every second for the next 120 seconds, Ctrl-C to stop...
using drive /media/tamhan/RPI-RP2/
fetch https://github.com/microsoft/devicescript-pico/releases/latest/download/devicescript-rp2040-pico.uf2
saved .devicescript/cache/devicescript-rp2040-pico.uf2 387072 bytes
cp .devicescript/cache/devicescript-rp2040-pico.uf2 /media/tamhan/RPI-RP2/
OK

```

Next, click the connect device option in the DeviceScript pane of VS Code. Nothing happens, then this error drops:
```
DeviceScript Development Server

using devs: v2.9.16, runtime: v2.9.16, node: v18.16.0 from /home/tamhan/devicescript/p1/DeviceScriptP1/node_modules/@devicescript/cli/built
devs: v2.9.16, runtime: v2.9.16, node: v18.16.0 running in /home/tamhan/devicescript/p1/DeviceScriptP1
services: error adding accelerometer, classIdentifier already in use
services: error adding brailledisplay, classIdentifier already in use
services: error adding devicescriptmanager, classIdentifier already in use
services: error adding distance, classIdentifier already in use
services: error adding ledstrip, classIdentifier already in use
services: error adding servo, classIdentifier already in use
services: error adding temperature, classIdentifier already in use
services: error adding vibrationmotor, classIdentifier already in use
   http     : http://127.0.0.1:8081
   websocket: ws://127.0.0.1:8081
   tcpsocket: tcp://127.0.0.1:8082
   dbgserver: tcp://127.0.0.1:8083
webclient: connected (ws1, 1 clients)
webclient: connected (ws2, 2 clients)
adding serial transport (requires "serialport" package)
{
  transport: 'serial',
  context: 'connect',
  exception: JDError [JacdacError]: JDUSB: can't connect, no HF2 nor JDUSB; port=/dev/ttyACM0, vendor: 2e8a, product: 420a
      at throwError (/home/tamhan/devicescript/p1/DeviceScriptP1/node_modules/@devicescript/cli/built/devicescript-cli.cjs:40566:13)
      at /home/tamhan/devicescript/p1/DeviceScriptP1/node_modules/@devicescript/cli/built/devicescript-cli.cjs:56472:7 {
    code: 'transport/hf2-not-supported',
    cancel: false
  }
}

```

The Raspberry Pi Pico device does show up well in the lsusb:
```
tamhan@TAMHAN18:~$ lsusb
Bus 007 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 014: ID 2e8a:420a  
Bus 001 Device 002: ID 1a40:0201 Terminus 

```

And also in the DMESG:
```

dmesg

[ 1565.340148] usb 1-1.5: new full-speed USB device number 15 using ehci-pci
[ 1565.455006] usb 1-1.5: New USB device found, idVendor=2e8a, idProduct=420a, bcdDevice= 1.00
[ 1565.455009] usb 1-1.5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1565.455012] usb 1-1.5: Product: RP2040
[ 1565.455013] usb 1-1.5: Manufacturer: Raspberry Pi
[ 1565.455015] usb 1-1.5: SerialNumber: JDe6605481db764e36
[ 1565.455954] cdc_acm 1-1.5:1.0: ttyACM0: USB ACM device
[ 1565.459044] input: Raspberry Pi RP2040 as /devices/pci0000:00/0000:00:12.2/usb1/1-1/1-1.5/1-1.5:1.2/0003:2E8A:420A.0018/input/input40
[ 1565.459311] hid-generic 0003:2E8A:420A.0018: input,hidraw2: USB HID v1.11 Gamepad [Raspberry Pi RP2040] on usb-0000:00:12.2-1.5/input2
[ 1565.461069] input: Raspberry Pi RP2040 as /devices/pci0000:00/0000:00:12.2/usb1/1-1/1-1.5/1-1.5:1.3/0003:2E8A:420A.0019/input/input41
[ 1565.520706] hid-generic 0003:2E8A:420A.0019: input,hidraw3: USB HID v1.11 Keyboard [Raspberry Pi RP2040] on usb-0000:00:12.2-1.5/input3
[ 1565.522329] input: Raspberry Pi RP2040 as /devices/pci0000:00/0000:00:12.2/usb1/1-1/1-1.5/1-1.5:1.4/0003:2E8A:420A.001A/input/input42
[ 1565.522556] hid-generic 0003:2E8A:420A.001A: input,hidraw4: USB HID v1.11 Mouse [Raspberry Pi RP2040] on usb-0000:00:12.2-1.5/input4
```

**Expected behavior**
This is really weird, as the system worked yesterday

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

pelikhan, 2023-06-04:
I suspect @mmoskal updated the C runtime to support 'fetch' and the VScode extension hasn't been updated yet.

We'll get this sorted out Monday.

pelikhan, 2023-06-04:
Our CLI should probably be able to take a version number; but it doesn't yet. 

https://microsoft.github.io/devicescript/api/cli#devs-flash

TAMHAN, 2023-06-04:
Hello Peli, thank you for your swift reply. Is there a way for me to get my hands onto the older build? I can flash the UF2 file myself!

TAMHAN, 2023-06-04:
Hello Peli, sorry, but bad news. I just tried the firmware from https://github.com/microsoft/devicescript-pico/releases and installed it by hand. This seems to be a two week old firmware...and does not work either. The error is the same.

TAMHAN, 2023-06-05:
Hello,
and an update. I also was able to connect to an ESP32-S2 module once. Now, I can no more:
{
  context: 'serial',
  exception: JDError [JacdacError]: serial device 303a:4001 (write error: ENODEV: no such device, write)
      at NodeWebSerialIO.error (/home/tamhan/devicescript/p1/DeviceScriptP1/node_modules/@devicescript/cli/built/devicescript-cli.cjs:57068:15)
      at /home/tamhan/devicescript/p1/DeviceScriptP1/node_modules/@devicescript/cli/built/devicescript-cli.cjs:57085:14
      at async /home/tamhan/devicescript/p1/DeviceScriptP1/node_modules/@devicescript/cli/built/devicescript-cli.cjs:56457:9 {
    code: undefined,
    cancel: false
  }
}

pelikhan, 2023-06-06:
We pushed a new firmware today.

Please update the extension to v2.10.2 and also update the dependencies in your project. Then flash the firmware again.

TAMHAN, 2023-06-06:
Hello,
thank you. I did the update, but no real avail on the RP2040. It now identifies as
"DEV> Raspberry Pi Pico v2.10.2; file v2.9.16
DEV> Raspberry Pi Pico v2.10.2; file v2.9.16"

When I plug it in, it shows up like so:
[  905.694598] usb 1-1.6: new full-speed USB device number 8 using ehci-pci
[  905.829482] usb 1-1.6: New USB device found, idVendor=2e8a, idProduct=420a, bcdDevice= 1.00
[  905.829487] usb 1-1.6: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  905.829490] usb 1-1.6: Product: RP2040
[  905.829492] usb 1-1.6: Manufacturer: Raspberry Pi
[  905.829495] usb 1-1.6: SerialNumber: JDe6605481db764e36
[  905.830449] cdc_acm 1-1.6:1.0: ttyACM0: USB ACM device
[  905.834795] input: Raspberry Pi RP2040 as /devices/pci0000:00/0000:00:12.2/usb1/1-1/1-1.6/1-1.6:1.2/0003:2E8A:420A.0009/input/input25
[  905.834951] hid-generic 0003:2E8A:420A.0009: input,hidraw2: USB HID v1.11 Gamepad [Raspberry Pi RP2040] on usb-0000:00:12.2-1.6/input2
[  905.836695] input: Raspberry Pi RP2040 as /devices/pci0000:00/0000:00:12.2/usb1/1-1/1-1.6/1-1.6:1.3/0003:2E8A:420A.000A/input/input26
[  905.895143] hid-generic 0003:2E8A:420A.000A: input,hidraw3: USB HID v1.11 Keyboard [Raspberry Pi RP2040] on usb-0000:00:12.2-1.6/input3
[  905.896762] input: Raspberry Pi RP2040 as /devices/pci0000:00/0000:00:12.2/usb1/1-1/1-1.6/1-1.6:1.4/0003:2E8A:420A.000B/input/input27
[  905.896960] hid-generic 0003:2E8A:420A.000B: input,hidraw4: USB HID v1.11 Mouse [Raspberry Pi RP2040] on usb-0000:00:12.2-1.6/input4

And it shows up for a second or so in VSCode (fully updated), only to disappear again afterward.

TAMHAN, 2023-06-06:
Hello,
and also tried with the ESP32-S2. Same issue - the board shortly shows up and disappears. It identifies like so:
DEV> Espressif ESP32-S2 (bare) v2.10.2; file v2.9.16
DEV> start: (no name) (null)

pelikhan, 2023-06-06:
```
DEV> Espressif ESP32-S2 (bare) v2.10.2; file v2.9.16
```

There is version mistmatch between the bytecode and the firmware runtime. Could you try to do "flash firmware..." again?

TAMHAN, 2023-06-06:
Hello,
thank you!
I tried to run this both on Windows and Linux, and always got this version combination. Interestingly, also on the ESP32 board...

pelikhan, 2023-06-06:
Actually it might be our "24h cache" messing around with you since we pushed the new binaries yesterday. If you use the CLI, try adding `--refresh`.

pelikhan, 2023-06-06:
Assuming you have this board (https://microsoft.github.io/devicescript/devices/esp32/feather-s2), run

```
devicescript flash --board feather_s2 --refresh
```

TAMHAN, 2023-06-06:
Hello,
ran it like so:
tamhan@TAMHAN18:~/devicescript/p1/DeviceScriptP1$  node ./node_modules/.bin/devicescript flash --board pico --install --refresh
using devs: v2.10.2, runtime: v2.10.2, node: v18.16.0 from /home/tamhan/devicescript/p1/DeviceScriptP1/node_modules/@devicescript/cli/built
no drives found; please plug your board while holding BOOTSEL button (or specify --drive)
Re-scanning every second for the next 120 seconds, Ctrl-C to stop...
using drive /media/tamhan/RPI-RP2/
fetch https://github.com/microsoft/devicescript-pico/releases/latest/download/devicescript-rp2040-pico.uf2
saved .devicescript/cache/devicescript-rp2040-pico.uf2 395264 bytes
cp .devicescript/cache/devicescript-rp2040-pico.uf2 /media/tamhan/RPI-RP2/
OK
tamhan@TAMHAN18:~/devicescript/p1/DeviceScriptP1$ 

Sadly, no progress:
DEV> Raspberry Pi Pico v2.10.2; file v2.9.16

And still the same issue!

pelikhan, 2023-06-06:
could you try to manually download and copy https://github.com/microsoft/devicescript-pico/releases/download/v2.10.200/devicescript-rp2040-pico.uf2 to your pico? It seems that we are deploying an older firmware.

pelikhan, 2023-06-06:
could you update the devicescript extension in vscode and run "DeviceScript: Report Issue..." . It will give us more info on what's happening (hopefully)

TAMHAN, 2023-06-06:
Hello, first the results of the YARN upgrade:
tamhan@TAMHAN18:~/devicescript/p1/DeviceScriptP1$ yarn upgrade @devicescript/cli@latest
yarn upgrade v1.21.1
warning ../../../package.json: No license field
[1/4] Resolving packages...
[2/4] Fetching packages...
[3/4] Linking dependencies...
[4/4] Rebuilding all packages...
success Saved lockfile.
success Saved 2 new dependencies.
info Direct dependencies
└─ @devicescript/cli@2.10.3
info All dependencies
├─ @devicescript/cli@2.10.3
└─ @devicescript/compiler@2.10.3
Done in 10.20s.

TAMHAN, 2023-06-06:
Hello,
second, I just ran the report issue function.

TAMHAN, 2023-06-06:
Hello,
and thirdly, I downloaded and manually deployed the UF2 file. Still, I get this result:
DEV> Raspberry Pi Pico v2.10.2; file v2.9.16

And still not working.

TAMHAN, 2023-06-06:
I can send you the file I just downloaded. The GitHub does not allow it attached. But,. here is the SHA1
tamhan@TAMHAN18:~/Downloads$ sha1sum "devicescript-rp2040-pico(1).uf2"
60e997dd8ff7efb6948fb5ae8d80370f1e185e3a  devicescript-rp2040-pico(1).uf2

It is different from an older file I downloaded some days ago, where I got this:
tamhan@TAMHAN18:~/Downloads$ sha1sum "devicescript-rp2040-pico.uf2"
af297e7185bbaf55fd7eb9051a08a7cffe29af16  devicescript-rp2040-pico.uf2

pelikhan, 2023-06-06:
@mmoskal i might be missing something obvious. Any idea?

TAMHAN, 2023-06-06:
Hello,
plase check the sha1sum if I use the right file. Or send me the file I should use by email. It is tamhan aeht tamoggemon dott com

TAMHAN, 2023-06-06:
Sadly not, the connection to the device is lost too quickly!

pelikhan, 2023-06-06:
Ha we might be getting somewhere. @mmoskal add the --clear flag to the flash cli command.

Try 

    devs flash --board pico --clear

TAMHAN, 2023-06-06:
Hello,
this worked:

tamhan@TAMHAN18:~/devicescript/p1/DeviceScriptP1$ node ./node_modules/.bin/devicescript flash --board pico --install --clear --refresh
error: unknown option '--clear'
(Did you mean --clean?)
tamhan@TAMHAN18:~/devicescript/p1/DeviceScriptP1$ node ./node_modules/.bin/devicescript flash --board pico --install --clean --refresh
using devs: v2.10.3, runtime: v2.10.3, node: v18.16.0 from /home/tamhan/devicescript/p1/DeviceScriptP1/node_modules/@devicescript/cli/built
no drives found; please plug your board while holding BOOTSEL button (or specify --drive)
Re-scanning every second for the next 120 seconds, Ctrl-C to stop...
using drive /media/tamhan/RPI-RP2/
fetch https://datasheets.raspberrypi.com/soft/flash_nuke.uf2
saved .devicescript/cache/flash_nuke.uf2 25600 bytes
cp .devicescript/cache/flash_nuke.uf2 /media/tamhan/RPI-RP2/
OK
tamhan@TAMHAN18:~/devicescript/p1/DeviceScriptP1$ node ./node_modules/.bin/devicescript flash --board pico --install --refresh
using devs: v2.10.3, runtime: v2.10.3, node: v18.16.0 from /home/tamhan/devicescript/p1/DeviceScriptP1/node_modules/@devicescript/cli/built
using drive /media/tamhan/RPI-RP2/
fetch https://github.com/microsoft/devicescript-pico/releases/latest/download/devicescript-rp2040-pico.uf2
saved .devicescript/cache/devicescript-rp2040-pico.uf2 395264 bytes
cp .devicescript/cache/devicescript-rp2040-pico.uf2 /media/tamhan/RPI-RP2/
OK
tamhan@TAMHAN18:~/devicescript/p1/DeviceScriptP1$ 

It now identifies like so:
DEV> Raspberry Pi Pico v2.10.2; file v2.10.0

Now I will experiment a bit more and report shortly!

Tam

TAMHAN, 2023-06-06:
Hello,
LOL, and I found something else. This program seems to kill the debug connection:
```

import { pins } from "@dsboard/pico"
import { startLightBulb } from "@devicescript/servers"

const led = startLightBulb({
    pin: pins.P10,
})

while(1===1){
    await led.intensity.write(1)
    await led.intensity.write(0)
    await led.intensity.write(1)
    await led.intensity.write(0)
}
```

pelikhan, 2023-06-06:
Closing this issue  as the root cause was found and the v2.10.4 will fix this.

closed 2023-06-06 by pelikhan

TAMHAN, 2023-06-06:
It was my honour thank you!

mmoskal, 2023-06-06:
For future reference: `DEV> Raspberry Pi Pico v2.10.2; file v2.9.16` - this means the firmware is 2.10.2 and the bytecode file (user program) is 2.9.16, which should be completely fine, except it was flooding the bus with too many packets. This is definitely something we need to fix, tracking as #432