microsoft/devicescript #468: Deploy cancelled - Device firmware mismatch

jaikamat, 2023-06-09:
**Describe the bug**

When attempting to deploy to the Pico using the VSCode extension, I am met with the error:

`DeviceScript - Deploy cancelled. Your device firmware (v2.10.9) is outdated (min v2.11.4).`

The Pico has been flashed with `v2.11.4`, so I'm wondering why the extension seems to detect an older version.

PS. Thanks team for all your hard work on this, very exciting

jaikamat, 2023-06-09:
Closing this issue as I see the new firmware version was [just released](https://github.com/microsoft/devicescript-pico/releases), thanks!

closed 2023-06-09 by jaikamat

pelikhan, 2023-06-09:
We forgot to bump the pico with the new screen support yesterday. Thanks for pointing it out.