microsoft/devicescript #474: Define shield in code and not .json

pelikhan, 2023-06-09:


mmoskal, 2023-06-23:
This seems to be supported. Here's an example:

```ts
import * as ds from "@devicescript/core"
import { ST7735Driver } from "@devicescript/drivers"
import { Image } from "@devicescript/graphics"
import {
    configureHardware,
    startBuzzer,
    startGamepad,
    startRelay,
} from "@devicescript/servers"
import { spi } from "@devicescript/spi"
import { pins as pins0 } from "@dsboard/pico"

/**
 * @devsWhenUsed
 */
export class ArcadeRP2040 {
    constructor() {
        configureHardware({
            log: {
                pinTX: pins0.P20,
            },
            jacdac: {
                pin: pins0.P11,
            },
        })
    }

    startGamepad() {
        return startGamepad({
            activeHigh: true,
            pinA: pins0.P2,
            pinB: pins0.P3,
            pinDown: pins0.P12,
            pinLeft: pins0.P13,
            pinMenu: pins0.P16,
            pinRight: pins0.P14,
            pinUp: pins0.P15,
        })
    }

    startBuzzer(): ds.Buzzer {
        return startBuzzer({ pin: pins0.P8, name: "speaker" })
    }

    startVibrationMotor(): ds.Relay {
        return startRelay({ pin: pins0.P4 })
    }

    async startDisplay() {
        spi.configure({
            mosi: pins0.P7,
            sck: pins0.P6,
            hz: 8_000_000,
        })

        pins0.P0.setMode(ds.GPIOMode.OutputHigh)

        const d = new ST7735Driver(Image.alloc(160, 128, 4), {
            dc: pins0.P9,
            cs: pins0.P5,
            reset: pins0.P10,
            frmctr1: 0x0e_14_ff,
            flip: true,
            spi: spi,
        })
        await d.init()
        return d
    }
}

interface BoardPins {
    /**
     * Pin P18 (GPIO18, io)
     *
     * @devsGPIO 18
     */
    P18: ds.IOPin

    /**
     * Pin P19 (GPIO19, io)
     *
     * @devsGPIO 19
     */
    P19: ds.IOPin

    /**
     * Pin P21 (GPIO21, io)
     *
     * @devsGPIO 21
     */
    P21: ds.IOPin

    /**
     * Pin GP1 (GPIO17, io) - broken out next to power switch
     *
     * @devsGPIO 17
     */
    GP1: ds.IOPin

    /**
     * Pin P26 (GPIO26, analogIn, io) - next to vibration motor
     *
     * @devsGPIO 26
     */
    ADC1: ds.AnalogInPin & ds.IOPin

    /**
     * Pin P27 (GPIO27, analogIn, io) - next to vibration motor
     *
     * @devsGPIO 27
     */
    ADC2: ds.AnalogInPin & ds.IOPin
}
export declare const pins: BoardPins
```

pelikhan, 2023-06-25:
Let’s chat about unifying how we define boards. It’s a mix of board.json or this kind of code oriented approach.

code looks better but the json does have some nice checks too.

import * as ds from ***@***.***/core"
import { ST7735Driver } from ***@***.***/drivers"
import { Image } from ***@***.***/graphics"
} from ***@***.***/servers"
import { spi } from ***@***.***/spi"
import { pins as pins0 } from ***@***.***/pico"

/**

pelikhan, 2023-06-27:
in many shield cases, the MCU can be different (e.g. pico vs pico-w) so we would have to have a different approach to define pins statically.

pelikhan, 2023-06-27:
It seems that this feature would require to be able to lookup pins by name in the static configuration of servers.

closed 2023-07-21 by pelikhan