microsoft/devicescript #492: Tutorial video shows invalid code (lightbulb no longer has `brightness`)

henrygab, 2023-06-19:
**Describe the bug**
Users following the tutorial video are instructed to create non-compiling code.
This will block users within the target audience.

**To Reproduce**
1. In browser, load https://github.com/microsoft/devicescript
2. Scroll down to embedded [video](https://user-images.githubusercontent.com/4175913/228997575-122d6ad0-f37e-4d4e-ab79-3d8f680ed99d.mp4).
3. Follow along in VSCode with video until ~1m56s, where video instructs to run the program under the debugger.

**Expected behavior**
Example / tutorial code is correct -- it must compile and execute.
The importance of having correct sample code cannot be overstated.

**Additional context**
As this is a video, it is not possible to provide a simple PR.

Problematic code:
```ts
import * as ds from "@devicescript/core"

const lightbulb = new ds.LightBulb();
setInterval(async() => {
    const brightness = await lightbulb.brightness.read();
    await lightbulb.brightness.write(brightness > 0 ? 0 : 1);
}, 1000) 
```

Likely correct code:
```ts
import * as ds from "@devicescript/core"

const lightbulb = new ds.LightBulb();
setInterval(async() => {
    // intensity is defined as a number between 0 and 0xFFFF
    const intensity = await lightbulb.intensity.read();
    console.log("intensity: ", intensity);
    if (intensity > 0) {
        await lightbulb.off();
    } else {
        await lightbulb.on();
    }
}, 1000) 
```

pelikhan, 2023-06-21:
Thank, video removed until we have the cycles to record it again

henrygab, 2023-06-21:
Ok.   Will be leaving issue open, as video removal does not appear to have been pushed yet.

henrygab, 2023-08-26:
Video still appears to show incorrect, non-compiling code, and remains in the readme.