microsoft/devicescript #499: Number functions

pelikhan, 2023-06-23:
More support for https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/Number/isInteger

ArchitGajjar, 2023-06-23:
HI @pelikhan I am interested to work on this issue. Can you please assign to me and  share more details about the use case and how we can test it ? Thanks

pelikhan, 2023-06-24:
I started a little guide at https://microsoft.github.io/devicescript/contributing#add-core-javascript-apis

Is this enough to get you started?

ArchitGajjar, 2023-07-04:
Yes, thank you!

pelikhan, 2023-08-15:
https://github.com/microsoft/devicescript/pull/608

PiotrKasperski, 2023-12-01:
If my PR #665 was good, I'm starting making rest

Marcjazz, 2024-01-24:
Hi @pelikhan, Can someone still worked this issue, I notice @PiotrKasperski has already done a lot of work on this, most especially with rest to the Number class static methods (#669). I will like to contribute with the Number class instance properties and methods implementation

pelikhan, 2024-01-24:
We are definitely more contributions! Always welcome. 🙏