microsoft/devicescript #503: Array js functions

pelikhan, 2023-06-24:
Finish Array object implementation.

https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/Array

Guide: https://microsoft.github.io/devicescript/contributing#add-core-javascript-apis

pelikhan, 2023-06-29:
Great!

- impl location: https://github.com/microsoft/devicescript/blob/main/packages/core/src/array.ts
- declarations: https://github.com/microsoft/devicescript/blob/main/packages/core/src/corelib.d.ts#L210
- tests: https://github.com/microsoft/devicescript/blob/main/devs/run-tests/24arraymap.ts

Feel free to break down implemetation in several PRs so that it helps the review.

sh33dafi, 2023-06-30:
Created a first PR https://github.com/microsoft/devicescript/pull/520