microsoft/devicescript #525: sample: weather display

pelikhan, 2023-07-01:
Write a new sample that uses a temperature/humidity sensor (e.g. SHT3, SHTC3, ....) and a character display (using ssd1306) to display the current temperature humidity.

guidelines on writing samples at https://microsoft.github.io/devicescript/contributing#samples

Bohemus307, 2023-07-08:
mind if i take this on as well?

pelikhan, 2023-07-08:
Yes please!

Bohemus307, 2023-07-09:
Hey @pelikhan, I got this working no problem in the simulator, but couldnt for the life of me get it to recognize any i2c devices attached to my Esp-32. It is an Adafruit feather huzzah wroom-32 but I didn't think that would matter as I am flashing it as a bare Esp-32. Have you had any luck getting either the sht30 or the ssd working outside the simulator or am I just missing something simple? Any ideas are appreciated thanks for your time on this.

pelikhan, 2023-07-09:
If you are starting from the esp32-bare configuration, you need to configure the i2c pins using `configureHardware`.... and I realize we don't have docs for this yet.

pelikhan, 2023-07-09:
Here is an example of using configureHardware: https://github.com/pelikhan/devicescript-pico-bricks/blob/main/src/index.ts#L25

Bohemus307, 2023-07-10:
Ok, @pelikhan I tried to add the board but ran into a bunch of issues with the pins colliding and getting errors with my build. So I backed out of the add board for now and have a pico coming in the mail for more efficient testing and just pushed a PR for the sample code and .mdx file. I would however like to dive into adding more boards and would love to understand the better pinout to use and how to handle pins colliding when they are multi-function on small boards like the huzzah (pinout here https://learn.adafruit.com/adafruit-huzzah32-esp32-feather/pinouts). Thanks so much for your help on thi, I'm really enjoying learning more about this project.

pelikhan, 2023-07-10:
You could also try to call "configureHardware" from code. It might be easier than configuring the board.json. You "just" need to configure the i2c pins in order to  enable the SHT driver.

We'll get it sorted out.

closed 2023-07-16 by pelikhan