microsoft/devicescript #559: SSD1306 I2C OLED 128x64 display not working with ESP32-DevKitC

varogonz95, 2023-07-16:
**Describe the bug**
I'm trying to make a SSD1306 display work with an ESP32 w/ USB Type C, this model specifically: 
[image]

I have connected the SDA and SCK pins to pins 21 and 22 respectively, as shown in this pinout diagram:
[image]

And set it up like shown in the docs:
```ts
import { SSD1306Driver } from "@devicescript/drivers"

const ssd = new SSD1306Driver({ 
    width: 128, 
    height: 64, 
    devAddr: 0x3d
})
await ssd.init()
ssd.image.print("Hello world!", 0, 0)
await ssd.show()
```

When I run the project, I get this:
```
 Unhandled exception
  message: I2C device not found or malfunctioning: I2C -1 dev=61: write [Buffer[2] 80ae], read undefined B
  at init_t main_F0 (pc:75)
```

**NOTE:** It works perfectly fine with Micropython, using this driver: https://github.com/micropython/micropython/blob/v1.19.1/drivers/display/ssd1306.py

**To Reproduce**
Steps to reproduce the behavior:
1. Connect a SSD1306 **SDA** and **SCK** pins to an ESP32 DevKit **21 and 22 pins respectively**.
2. Copy/Paste the code from the docs example for SSD1306, and set `width`, `height` and `devAddr` setting accordingly.
3. Connect the ESP32 device
4. Run the project

**Expected behavior**
To see a "Hello world" text in the SSD1306 display.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

pelikhan, 2023-07-16:
I suspect this is due to the missing import for the devkitc pin configuration -- see https://microsoft.github.io/devicescript/devices/esp32/esp32-devkit-c#devicescript-import .

I've added a little faq about this error message (https://microsoft.github.io/devicescript/developer/errors#i2c-device-not-found-or-malfunctioning) that will hopefully help in the future.

pelikhan, 2023-07-16:
The esp32-devkit-c does not have an i2c configuration, so also follow the instructions at https://microsoft.github.io/devicescript/developer/board-configuration#configuration-by-code to configure the i2c pins.

varogonz95, 2023-07-16:
I followed all sections mentioned:
1. I added the import for devkitc
2. I created a new board config with I2C setup for pins 21 and 22
3. I set up the I2C in-code
and still no luck, same error message.

I can confirm this works with micropyhton

pelikhan, 2023-07-16:
Could you share your project sources?

pelikhan, 2023-07-16:
I just release v2.14.2 which has an improved flow to collect a bug report from Visual Studio Code. After updating the extension, run "DeviceScript: Report Issue".

varogonz95, 2023-07-21:
@pelikhan Sorry for the late response.
You can find the sources at this repo: https://github.com/varogonz95/devicescript-pgchi.git

varogonz95, 2023-08-13:
@pelikhan any new on this? Im willing to contribute if necessary

pelikhan, 2023-08-14:
could you carve out a minimal repro/snippet?

varogonz95, 2023-08-15:
Sure, this is what I have, in a nutshell:
```ts
import { configureHardware, startLightLevel, startSoilMoisture } from "@devicescript/servers";
import { pins } from "@dsboard/esp32_wroom_devkit_c";

configureHardware({
    i2c: {
        pinSDA: pins.P21,
        pinSCL: pins.P22
    }
})

const ssdDisplay = new SSD1306Driver({ height: 64, width: 128 })

try {
     const characterScreen = await startCharacterScreen(ssdDisplay)
     await characterScreen.message.write("Hello world!")
} catch (error) {
    console.warn(`${SSD1306Driver.constructor.name} is not available:`, error)
}

```

varogonz95, 2023-08-16:
vscode ask me to upgrade firmware to 2.15.1, which I already did, and it wont let me run anything until I do so :(

pelikhan, 2023-08-16:
Oh right, please try again.

varogonz95, 2023-08-17:
Unfortunately that didnt work :(
Output:
```
2023-08-16 22:37:25.123 [warning] dev:  Espressif ESP32-DevKitC v2.15.1; file v2.15.1
2023-08-16 22:37:25.162 [warning] dev:  start: (no name) v0.0.1-9ad74c3-2023.08.17.04.37
2023-08-16 22:37:25.191 [error] dev:  Unhandled exception
2023-08-16 22:37:25.205 [warning] dev:  Exception: DriverError
2023-08-16 22:37:25.205 [warning] dev:   message: I2C device not found or malfunctioning: I2C error -1 dev=61: write [Buffer[2] 80ae], read undefined B
2023-08-16 22:37:25.205 [warning] dev:   at init_F52 (pc:52)
2023-08-16 22:37:25.216 [warning] dev:   at startCharacterScreen_F11 (pc:24)
2023-08-16 22:37:25.219 [warning] dev:   at main_F0 (pc:515)
2023-08-16 22:37:25.220 [error] dev:  Unhandled exception
2023-08-16 22:37:25.225 [warning] dev:  stop program
```

varogonz95, 2023-08-17:
WAIT. It works now!
I had to change the address tho.
Evidence:
[image]

pelikhan, 2023-08-17:
Victory! i2c adresses....

closed 2023-10-04 by varogonz95