microsoft/devicescript #568: add ILI9341 320x240 LCD driver

mmoskal, 2023-07-20:
The ILI9341 driver should be the same as ST7735 except for the init sequence and screen size. An example init sequence is here https://github.com/adafruit/Adafruit_ILI9341/blob/master/Adafruit_ILI9341.cpp#L151

A new class needs to be added similar to this https://github.com/microsoft/devicescript/blob/9cd49aef06114a147cbd5127fc12f1fc58112769/packages/drivers/src/st7735.ts#L247-L264

The main thing is to test it with hardware

** Datasheet, firmware, etc... **

https://cdn-shop.adafruit.com/datasheets/ILI9341.pdf

aeroniemi, 2023-11-21:
This works as a basic implementation:

```ts
export interface ILI934xOptions extends STLikeDisplayOptions { }

export class ILI934xDriver extends STLikeDisplayDriver {
    constructor(image: Image, options: ILI934xOptions) {
        super(
            image,
            options,
            hex`
01 80 80 // software reset
ED 04 64 03 12 81 //power on sequence control
3A 01 55 // pixel format
B1 02 00 18 // FRMCTR1
B6 03 08 A2 27 // display function control
11 80 78
29 80 78
`
        )
    }
}
```
Tested on [an adafruit 320x240 ILI9340C](https://www.adafruit.com/product/1480)
it'll probably be worth adding more options but i'm currently unsure what exactly would be useful (ie what varies between products)

pelikhan, 2023-11-21:
https://github.com/microsoft/devicescript/pull/663

pelikhan, 2023-11-22:
Added!

closed 2023-11-26 by pelikhan