microsoft/devicescript #602: Development Gateway

louisvangeldrop, 2023-08-05:
I have added a development gateway at a Raspberrypi with the following connection:   http://192.168.0.138:7071 in order to send some MQTT messages to Node-Red.

Since MQTT-client is supported now, I don't need that gateway anymore for that purpose and stopped it. 
The extension is however still trying to connect to the gateway. I can't find a command to delete the existing entry in "Development Gateway" tab. Any suggestion?

The following error message is showing every second in the Terminal window: WSSK-H: connecting to ws://192.168.0.138:7071/wssk/main/fe34851805884cff

pelikhan, 2023-08-06:
For now, try to press connect and enter to have an empty connection string.

louisvangeldrop, 2023-08-06:
That works. Thx for quick response