microsoft/devicescript #610: A fatal error occurred: Failed to write to target RAM (result was 01070000: Operation timed out)

zhengAnBin, 2023-08-17:
**Describe the bug**
on the M1 platform, Flash Firmware -> esp32-c3
Appear:
```
(base) macos@macos test % node ./node_modules/.bin/devicescript flash --board esp32c3_bare --refresh --instal
l
using devs: v2.15.1, runtime: v2.15.1, node: v18.12.1 from /Users/macos/hardware/test/node_modules/@devicescript/cli/built
using serial port /dev/cu.usbmodem56910241431 at 1500000
esptool: /Users/macos/miniforge3/bin/esptool.py
fetch https://github.com/microsoft/devicescript-esp32/releases/latest/download/devicescript-esp32c3-esp32c3_bare-0x0.bin
(node:3153) ExperimentalWarning: The Fetch API is an experimental feature. This feature could change at any time
(Use `node --trace-warnings ...` to show where the warning was created)
saved .devicescript/cache/devicescript-esp32c3-esp32c3_bare-0x0.bin 1282048 bytes
run: /Users/macos/miniforge3/bin/esptool.py --port /dev/cu.usbmodem56910241431 --baud 1500000 write_flash 0x0 .devicescript/cache/devicescript-esp32c3-esp32c3_bare-0x0.bin
esptool.py v4.6.2
Serial port /dev/cu.usbmodem56910241431
Connecting....
Detecting chip type... ESP32-C3
Chip is ESP32-C3 (revision v0.4)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 48:31:b7:34:9c:54
Uploading stub...

A fatal error occurred: Failed to write to target RAM (result was 01070000: Operation timed out)
flash faile
```

zhengAnBin, 2023-08-17:
It's not a problem with devicescript.

pelikhan, 2023-08-17:
Can we close this issue?

zhengAnBin, 2023-08-21:
> Can we close this issue?

yes

closed 2023-08-21 by pelikhan