microsoft/devicescript #611: `devicescript devtools --vscode` terminated with exit code: 100

xmeow, 2023-08-18:
**Describe the bug**
The devicescript service failed to start, please see the screen shoot. And hardware connection is not available. Vscode will stuck at the context, and the only way to resolve is to kill all vscode process. 

**To Reproduce**
When try to click the button 'connect device'. 

**Screenshots**
[image]

**Additional context**
From the electron devtools it says that the **bus** is undefined at the codeline:
`this.extensionState.bus.setCustomServiceSpecifications`

Nasreddine, 2023-10-22:
check if following ports are not used by other processes.
 http     : http://localhost:8081
   websocket: ws://localhost:8081
   tcpsocket: tcp://localhost:8082
   dbgserver: tcp://localhost:8083

pelikhan, 2023-10-25:
we should be sniffing for unused ports...

pelikhan, 2023-11-06:
@xmeow we had an unneeded dependency on serialport. is this still failing?

pelikhan, 2023-11-07:
> It looks decent on latest devicescript extension build.
>
> You are receiving this email because you commented on the thread.

[like]  Peli de Halleux reacted to your message:

reggi, 2024-01-31:
I found that I ran into this when I'd run "Connect Device" in vs code, then try to hit the play button on the debugger.

The workflow is much better to run `npm run start`, and then just save your main file to upload it.