microsoft/devicescript #612: Add ESP32-C3FH4-RGB

gsteiert, 2023-08-24:
Add support for ESP32-C3FH4-RGB 
https://github.com/01Space/ESP32-C3FH4-RGB

The attached file should take care of it if you remove the .txt suffix

**Additional context**
[esp32_c3fh4_rgb.board.json.txt](https://github.com/microsoft/devicescript/files/12424967/esp32_c3fh4_rgb.board.json.txt)

pelikhan, 2023-08-24:
@mmoskal let's chat about bringing this in.

gsteiert, 2023-09-09:
Any status on this?  Do you need help with the board page?

pelikhan, 2023-09-11:
It slipped through the cracs. We'll take a look soon.

pelikhan, 2023-09-11:
Could you take a picture of this device using this template and attach it to the bug? 

https://microsoft.github.io/jacdac-docs/templates/jacdac-device-photo-template.pdf

gsteiert, 2023-09-12:
Will this work?
[image]

gsteiert, 2023-09-12:
this might be a little better
[image]

pelikhan, 2023-09-12:
https://github.com/microsoft/jacdac/pull/1329

mmoskal, 2023-09-12:
should be fixed now (once everything builds)

closed 2023-09-12 by mmoskal

gsteiert, 2023-09-13:
Thanks for adding the board.
It might be nice to mention the 5x5 LED grid is connected to LEDS pin GPIO8 on the board page.
This makes a good board to demo the LED driver.

[image]

pelikhan, 2023-09-13:
Right we need to support having a bit of typescript in the board definition.

pelikhan, 2023-09-13:
example added https://github.com/microsoft/devicescript/commit/ca5fa26485df3addbb2d7225ecb8f08dccfb9135