microsoft/devicescript #624: Infinite loop in Array.sort

louisvangeldrop, 2023-09-17:
1. Create an arbitrary Array of numbers with length>100
2. myArray.sort() results in:
3.  Exception: InfiniteLoop
VM>  at sort [as F9_pc84] (node_modules\@devicescript\core\src\array.ts:206:13)
VM>  at main [as F0_pc37] (src\main.ts:90:1)

mmoskal, 2023-11-01:
I suspect it just runs for too long - the algorithm is quadratic - PRs are welcome!

We limit execution to `DEVS_MAX_STEPS` (128k) opcodes before you need to yield.

louisvangeldrop, 2023-11-01:
Is the DEVS_MAX_STEPS a parameter that one can define during runtime?

mmoskal, 2023-11-01:
Not currently.

IvanMtze, 2023-12-22:
@mmoskal Have you consider using timsort algorithm instead of the current implementation? I believe is what V8 uses. I can work on this if that is a better approach than the code has for now. :)

mmoskal, 2023-12-22:
It would be most welcome!