microsoft/devicescript #647: M5Stack support

deboboy, 2023-10-19:
Hola,
Does DeviceScript support M5Stack?

https://www.m5stack.com/

If yes then I’ll buy one of their devices and continue experimenting with your VS Code plugin.

pelikhan, 2023-10-19:
Modules with the supported ESP32-c3, esp32-s3 should work as long as we have access to the native usb port. Some breakout introduce a UART chip that is slower/buggy.

What model in particular you had in mind?

As with all hardware projects, we would need to test this with a physical device to have a definite answer.

deboboy, 2023-10-19:
Looking at this M5Stack model; to experiment with wearable scenarios:
https://www.adafruit.com/product/4289

pelikhan, 2023-10-22:
This one has a UART interface chip so it is likely that devicescript does not work well with it.