microsoft/devicescript #674: RGB Pins occupied by PSRAM

hypercreeper, 2023-12-27:
**Describe the bug**
My board uses pins 16, 4, 17 for builtin rgb led, but everytime i try to make a custom board for it, it keeps telling me that the pins are used by psram. 

**To Reproduce**
Create a new board.json with this code in it: 
```
"led": {
        "rgb": [
            {"pin": "LED0"},
            {"pin": "LED1"},
            {"pin": "LED2"}
        ]
    },
```
and run `npm run build`
**Expected behavior**
The system should build succesfully and I can flash the firmware to my board.
**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
The board is an ESP32-2432S028R.
This is the output of `npm run build`: 
```
> testno2@0.0.0 build
> npm run build:devicescript

> testno2@0.0.0 build:devicescript
> devicescript build src/main.ts

using devs: v2.15.23, runtime: v2.15.23, node: v18.7.0 from C:\Users\hyper\Desktop\DeviceScript\testno2\testno2\node_modules\@devicescript\cli\built
C:\Users\hyper\Desktop\DeviceScript\testno2\testno2\boards\esp32_2432s028r.board.json(1): LED0=17 has 'psram' function
LED2=16 has 'psram' function
```
board.json file: 
```
{
    "$schema": "https://raw.githubusercontent.com/microsoft/devicescript-esp32/main/boards/esp32deviceconfig.schema.json",
    "devName": "ESP32-2432S028R",
    "productId": "0x3462c836",
    "$description": "Bare ESP32 without any default functions for pins.",
    "archId": "esp32",
    "url": "https://www.espressif.com/en/products/socs/esp32",
    "pins": {
        "P35": 35,
        "P22": 22,
        "P21": 21,
        "P27": 27,
        "LED0": 17,
        "LED1": 4,
        "LED2": 16
    },
    "led": {
        "rgb": [
            {"pin": "LED0"},
            {"pin": "LED1"},
            {"pin": "LED2"}
        ]
    },
    "sd": {
        "pinCS": 5,
        "pinMISO": 19,
        "pinMOSI": 23,
        "pinSCK": 18
    }
}
```

pelikhan, 2024-01-02:
@mmoskal any idea?

mmoskal, 2024-01-02:
Right, 16 and 17 are used by PSRAM if PSRAM is installed. I assume your board doesn't have it. I guess we should have an option in the .board.json to disable PSRAM checking, however the board is unlikely to work very well anyways due to lack of a native USB interface.

From https://microsoft.github.io/devicescript/devices#implementation-status

The ESP32-C3 boards are best supported. The regular ESP32 (without -C3 or -S2) currently have issues with the USB connection (as it's handled by external chip). The ESP32-S2 has limited memory which makes it difficult to use TLS. The ESP32-S3 is very recent and largely untested.

The RP2040 should generally work, but TLS is not supported on Pico-W.