microsoft/devicescript #676: fetch is not defined

toohandsome, 2024-01-01:
## Describe the issue
A clear and concise description of what the bug is.

## To Reproduce
when i select flash , it will show a error 

```
using serial port COM3 at 1500000
esptool: C:\Users\Administrator\.espressif\python_env\idf5.1_py3.11_env\Scripts\esptool.py
fetch https://github.com/microsoft/devicescript-esp32/releases/latest/download/devicescript-esp32c3-esp32c3_bare-0x0.bin
F:\esp32\node_modules\@devicescript\cli\built\devicescript-cli.cjs:490875
    const resp = await fetch(dlUrl);
                 ^

ReferenceError: fetch is not defined
    at fetchFW (F:\esp32\node_modules\@devicescript\cli\built\devicescript-cli.cjs:490875:18)
    at async flashESP32 (F:\esp32\node_modules\@devicescript\cli\built\devicescript-cli.cjs:490730:21)
```

## Expected behavior
A clear and concise description of what you expected to happen.

## Environment

vscode: 1.85.1
extension: 2.15.23
devsVersion: v2.15.23
runtimeVersion: v2.15.23
nodeVersion: v16.17.0

## Sources
-  src/main.ts}
```ts title="src/main.ts"
import * as ds from "@devicescript/core"

setInterval(async () => {
    console.log(":)")
}, 1000)

```

## Devices

interaction_mode: 0
auto_connect: false
reset_in: false
last_reset_in_time: 0
transport:
  web: connected
device: 
  id: VX12 (0xca7adf6c60073afc)
  product: ? (0x?)
  firmware_version: 
  uptime: 
  stats: announce 461, drop 0, restart 0
  services:
    infrastructure (0x1e1589eb)
    proxy (0x16f19949) 

**Screenshots**
[image]

toohandsome, 2024-01-01:
I konw it's why , node version need 18+; but document  is require16+....
i update node version to 20, it's solved.
but i have a new problem.

```
PS F:\esp32> node_modules\.bin\devicescript.cmd flash --board esp32c3_bare --refresh --install
using devs: v2.15.23, runtime: v2.15.23, node: v20.10.0 from F:\esp32\node_modules\@devicescript\cli\built
using serial port COM3 at 1500000
esptool: C:\Users\Administrator\.espressif\python_env\idf5.1_py3.11_env\Scripts\esptool.py
fetch https://github.com/microsoft/devicescript-esp32/releases/latest/download/devicescript-esp32c3-esp32c3_bare-0x0.bin
saved .devicescript/cache/devicescript-esp32c3-esp32c3_bare-0x0.bin 1282048 bytes
run: C:\Users\Administrator\.espressif\python_env\idf5.1_py3.11_env\Scripts\esptool.py --port COM3 --baud 1500000 write_flash 0x0 .devicescript/cache/devicescript-esp32c3-esp32c3_bare-0x0.bin
node:internal/errors:563
    ErrorCaptureStackTrace(err);
    ^

Error: spawn UNKNOWN
    at ChildProcess.spawn (node:internal/child_process:421:11)
    at spawn (node:child_process:761:9)
    at F:\esp32\node_modules\@devicescript\cli\built\devicescript-cli.cjs:490903:49
    at new Promise (<anonymous>)
    at runTool (F:\esp32\node_modules\@devicescript\cli\built\devicescript-cli.cjs:490895:10)
    at runEsptool (F:\esp32\node_modules\@devicescript\cli\built\devicescript-cli.cjs:490803:23)
    at flashESP32 (F:\esp32\node_modules\@devicescript\cli\built\devicescript-cli.cjs:490745:28) {
  errno: -4094,
  code: 'UNKNOWN',
  syscall: 'spawn'
}

Node.js v20.10.0
```

pelikhan, 2024-01-02:
I've pushed a new version (2.16.0) that should require node v20+. 

Do you have esptool installed  (see https://microsoft.github.io/devicescript/devices/esp32#requirements )?

toohandsome, 2024-01-03:
> I've pushed a new version (2.16.0) that should require node v20+.
> 
> Do you have esptool installed (see https://microsoft.github.io/devicescript/devices/esp32#requirements )?

yes , I have installed esptool

[image]

pelikhan, 2024-02-01:
it seems that we are launching `esptool.py` instead of `esptool`. were you able to flash manually?