microsoft/devicescript #678: Device firmware is outdated, but no newer firmware is available

343max, 2024-01-07:
Type: <b>Bug</b>

## Describe the issue

When trying to run sample code on my ESP32 C3 Super Mini I receive this error:

DeviceScript - Deploy cancelled. Your device firmware (v2.15.15) is outdated (min v2.16.0).

The dialog offers me to update the firmaware but will only update to exactly the same version. The latest firmware on [https://github.com/microsoft/devicescript-esp32/releases](the release page) is 2.15.1500 with no 2.16.0 firmware available.

## To Reproduce
- create a new empty project
- install the firmware on a esp32 c3 using `npx devicescript flash --board esp32c3_supermini`
- connect to the device in vscode
- try to run a script by clicking the "DeviceScript: Debug" in the tab bar of the main.ts file

## Expected behavior
Something should be uploaded to the esp32 and be running.

## Environment

vscode: 1.85.1
extension: 2.16.0
devsVersion: v2.16.0
runtimeVersion: v2.16.0
nodeVersion: v21.5.0
python: 3.9.6

## Devices

interaction_mode: 0
auto_connect: false
reset_in: false
last_reset_in_time: 0
transport:
  web: connected
device: 
  id: XY30 (0x1b39855c84b943e8)
  product: ? (0x?)
  firmware_version: 
  uptime: 
  stats: announce 434, drop 0, restart 0
  services:
    infrastructure (0x1e1589eb)
    dashboard (0x1be59107)

device: 
  id: XW93 (0x2ca58fdedb600877)
  product: ? (0x?)
  firmware_version: 
  uptime: 
  stats: announce 436, drop 0, restart 0
  services:
    infrastructure (0x1e1589eb)
    proxy (0x16f19949)

device: 
  id: HH86 (0x78299d06f8e9b850)
  product: ? (0x?)
  firmware_version: 
  uptime: 
  stats: announce 357, drop 0, restart 0
  services:
    infrastructure (0x1e1589eb)
    dashboard (0x1be59107)

device: 
  id: KA66 (0xfeecda3b3a9af4ff)
  product: ? (0x31606c1c)
  firmware_version: v2.15.1500
  uptime: 193833.15895801736
  stats: announce 377, drop 0, restart 0
  services:
    roleManager (0x1e4b7e66)
        rw auto_bind: true (01) 
            last data: 218641, last get: 218641, last set: , last gets attempts: 0,
        ro all_roles_allocated: true (01) 
            last data: 216062, last get: 216062, last set: , last gets attempts: 0,
        event change: 1
    deviceScriptManager (0x1134ea2b)
        ro status_code: 0x4,0x0 (04000000) 
            last data: 216943, last get: 216943, last set: , last gets attempts: 0,
        rw running: false (00) 
            last data: 216424, last get: 216424, last set: , last gets attempts: 0,
        rw autostart: false (00) 
            last data: 216006, last get: 216006, last set: 165128, last gets attempts: 0,
        ro program_size: 0x0 (00000000) 
            last data: 216062, last get: 216062, last set: , last gets attempts: 0,
        ro program_hash: 0x0 (00000000) 
            last data: 216062, last get: 216062, last set: , last gets attempts: 0,
        ro program_sha256: 0000000000000000000000000000000000000000000000000000000000000000 | ................................ (0000000000000000000000000000000000000000000000000000000000000000) 
            last data: 216062, last get: 216062, last set: , last gets attempts: 0,
        const runtime_version: 15.0.15,0xf,0x2 (0f000f02) 
            last data: 165135, last get: 165135, last set: , last gets attempts: 0,
        ro program_name:  () 
            last data: 216062, last get: 216062, last set: , last gets attempts: 0,
        ro program_version:  () 
            last data: 216062, last get: 216062, last set: , last gets attempts: 0,
        event status_code_changed: 1
    devsDbg (0x155b5b40)
        rw enabled: true (01) 
            last data: 216006, last get: 216006, last set: , last gets attempts: 0,
    settings (0x1107dc4a)
    wifi (0x18aae1fa)
        ro rssi: -128dB (80) 
            last data: 31137, last get: 31137, last set: , last gets attempts: 0,
        rw enabled: true (01) 
            last data: 216062, last get: 216062, last set: , last gets attempts: 0,
        ro ip_address: 192.168.178.63 (c0a8b23f) 
            last data: 217226, last get: 217226, last set: , last gets attempts: 0,
        const eui_48: ec:da:3b:3a:9a:f4 (ecda3b3a9af4) 
            last data: 41937, last get: 41937, last set: , last gets attempts: 0,
        ro ssid: "343max" (3334336d6178) 
            last data: 217294, last get: 217294, last set: , last gets attempts: 0,
        event got_ip: 1
        event lost_ip: 1
        event scan_complete: 7
        event networks_changed: 1
    cloudConfiguration (0x1462eefc)
        ro server_name:  () 
            last data: 218642, last get: 218642, last set: , last gets attempts: 0,
        ro cloud_device_id:  () 
            last data: 218642, last get: 218642, last set: , last gets attempts: 0,
        const cloud_type: "WSSK" (5753534b) 
            last data: 41903, last get: 41903, last set: , last gets attempts: 0,
        ro connection_status: Disconnected (0200) 
            last data: 218642, last get: 218642, last set: , last gets attempts: 0,
    cloudAdapter (0x14606e9c)
        ro connected: false (00) 
            last data: 217294, last get: 217294, last set: , last gets attempts: 0,
        ro connection_name: "(null) (WSSK)" (286e756c6c2920285753534b29) 
            last data: 218072, last get: 218072, last set: , last gets attempts: 0,

Extension version: 2.16.0
VS Code version: Code 1.85.1 (Universal) (0ee08df0cf4527e40edc9aa28f4b5bd38bbff2b2, 2023-12-13T09:48:06.308Z)
OS version: Darwin arm64 23.2.0
Modes:

<details>
<summary>System Info</summary>

|Item|Value|
|---|---|
|CPUs|Apple M2 (8 x 24)|
|GPU Status|2d_canvas: enabled<br>canvas_oop_rasterization: enabled_on<br>direct_rendering_display_compositor: disabled_off_ok<br>gpu_compositing: enabled<br>multiple_raster_threads: enabled_on<br>opengl: enabled_on<br>rasterization: enabled<br>raw_draw: disabled_off_ok<br>video_decode: enabled<br>video_encode: enabled<br>vulkan: disabled_off<br>webgl: enabled<br>webgl2: enabled<br>webgpu: enabled|
|Load (avg)|3, 5, 5|
|Memory (System)|24.00GB (0.36GB free)|
|Process Argv|--crash-reporter-id 1321f563-b9bb-4697-a383-a4c8545a891f|
|Screen Reader|no|
|VM|0%|
</details><details>
<summary>A/B Experiments</summary>

```
vsliv368cf:30146710
vsreu685:30147344
vspor879:30202332
vspor708:30202333
vspor363:30204092
vslsvsres303:30308271
vshan820:30294714
vscorecescf:30445987
vscod805:30301674
binariesv615:30325510
bridge0708:30335490
bridge0723:30353136
vsaa593cf:30376535
py29gd2263:30899288
vsclangdc:30486549
c4g48928:30535728
azure-dev_surveyone:30548225
f6dab269:30613381
f8hc8238:30694864
pythongtdpath:30769146
i26e3531:30792625
welcomedialog:30910333
pythonidxpt:30866567
pythonnoceb:30805159
asynctok:30898717
pythontestfixt:30902429
pythonregdiag2cf:30928864
pyreplss1:30897532
pythonmypyd1:30879173
pythoncet0:30885854
pythontbext0:30879054
accentitlementsc:30887149
dsvsc016:30899300
dsvsc017:30899301
dsvsc018:30899302
aa_t_chat:30882232
dsvsc019cf:30917260
cp7184t3:30927821

```

</details>

pelikhan, 2024-01-07:
Oopsies, we'll trigger firmware builds tomorrow. Meanwhile there is a user setting in vscode to ignore those version checks.

pelikhan, 2024-01-07:
Fixed thanks for reporting!

closed 2024-01-07 by pelikhan

343max, 2024-01-07:
Now it's working as expected. Thank you!