microsoft/devicescript #679: M5Stack Cardputer pins not exposed

reggi, 2024-01-31:
I'm trying to get the [m5stack cardputer](https://shop.m5stack.com/products/m5stack-cardputer-kit-w-m5stamps3) running with device script. It has a `ESP32-S3 chip` and a [bunch of pins](https://docs.m5stack.com/en/api/core/gpio), I'm specifically running into this issue here, while trying to get the screen working via the `ST7789Driver`, I'm trying to [reverse engineer this c++ code](https://github.com/m5stack/M5GFX/blob/master/src/M5GFX.cpp#L1344-L1387) of the current pin configuration for this LCD

```ts
DEV: *  message: pin 35 not exposed; available: ["P11","P10","P12","P13","P17","P15","P16","P14","P2","P3","P1","P7","P4","P6","P5","P8","P9","P0... 
```

This is my code:

```
import * as ds from "@devicescript/core"
import { spi } from "@devicescript/spi"
import { gpio } from "@devicescript/core"
import { ST7789Driver } from "@devicescript/drivers"
import { Image } from "@devicescript/graphics"

spi.configure({
    mosi: gpio(35),
    sck: gpio(36),
    hz: 8_000_000,
})

// backlight led
gpio(32).setMode(ds.GPIOMode.OutputHigh)

const display = new ST7789Driver(Image.alloc(240, 135, 4), {
    dc: gpio(34),
    cs: gpio(37),
    reset: gpio(33),
    // frmctr1: 0x0e_14_ff,
    flip: false,
    spi: spi,
    offX: 52,
    offY: 40,
})
await display.init()
display.image.print("Hello world!", 3, 10)
await display.show()
```

In order to get my device working I flashed it with this board config [esp32s3_devkit_m](https://github.com/microsoft/devicescript-esp32/blob/main/boards/esp32s3/esp32s3_devkit_m.board.json) but pin 35 isn't on here, in fact many pins:

| pin No | Name | allocation                                   |
| ------ | ---- | -------------------------------------------- |
| 0      | G0   | downloader                                   |
| 1      | T1   | UART                                         |
| 2      | G2   | Side terminal (except M5FIRE),M5-BUS         |
| 3      | R1   | UART                                         |
| 4      | G4   | TF                                           |
| 5      | G5   | Side terminal (except M5FIRE),M5-BUS         |
| 6      | G6   | SDIO                                         |
| 7      | G7   | SDIO                                         |
| 8      | G8   | SDIO                                         |
| 9      | G9   | SDIO                                         |
| 10     | G10  | SDIO                                         |
| 11     | G11  | SDIO                                         |
| 12     | G12  | IIS_SCLK                                     |
| 13     | G13  | IIS_WS                                       |
| 14     | G14  | LCD                                          |
| 15     | G15  | IIS_OUT                                      |
| 16     | R2   | UART                                         |
| 17     | T2   | UART                                         |
| 18     | G18  | TF,Top terminal(SCK),M5-BUS                  |
| 19     | G19  | TF,Top terminal(MISO),M5-BUS                 |
| 21     | G21  | GROVE-A(SDA)                                 |
| 22     | G22  | GROVE-A(SCL)                                 |
| 23     | G23  | TF,Top terminal(MOSI)                        |
| 25     | G25  | Speaker,Side terminal (except M5FIRE),M5-BUS |
| 26     | G26  | Side terminal (except M5FIRE),M5-BUS         |
| 27     | G27  | LCD                                          |
| 32     | G32  | LCD BackLight                                |
| 33     | G33  | LCD                                          |
| 34     | G34  | None                                         |
| 35     | G35  | Side terminal (except M5FIRE)                |
| 36     | G36  | Side terminal (except M5FIRE)                |
| 37     | G37  | Button C                                     |
| 38     | G38  | Button B                                     |
| 39     | G39  | Button A                                     |

Creating a custom board with pin 35 set is giving me this error, also happens when I try to flash :

> /Users/thomasreggi/Desktop/devicescript/boards/cardputer.board.json(1): P35=35 has 'flash' function

```
➜  devicescript npx devicescript flash esp32 --board cardputer --verbose
verbose level: 1
using devs: v2.16.1, runtime: v2.16.1, node: v18.17.1 from /Users/thomasreggi/Desktop/devicescript/node_modules/@devicescript/cli/built
build config from: .
compile: devicescript 
P0: GPIO0, boot, io
P1: GPIO1, analogIn, io
P10: GPIO10, analogIn, io
P11: GPIO11, io
P12: GPIO12, io
P13: GPIO13, io
P14: GPIO14, io
P15: GPIO15, io
P16: GPIO16, io
P17: GPIO17, analogOut, io
P18: GPIO18, analogOut, io
P2: GPIO2, analogIn, io
P21: GPIO21, io
P3: GPIO3, analogIn, io
P33: GPIO33, io
P34: GPIO34, io
P35: GPIO35, flash, io
P38: GPIO38, io
P39: GPIO39, debug, io
P4: GPIO4, analogIn, io
P40: GPIO40, debug, io
P41: GPIO41, debug, io
P42: GPIO42, debug, io
P43: GPIO43, log.pinTX, io
P45: GPIO45, boot, io
P46: GPIO46, boot, io
P47: GPIO47, io
P48: GPIO48, led.pin, io
P5: GPIO5, analogIn, io
P6: GPIO6, analogIn, io
P7: GPIO7, analogIn, io
P8: GPIO8, analogIn, io
P9: GPIO9, analogIn, io
boards/cardputer.board.json(1): P35=35 has 'flash' function
Please select board, available options:
  ESP32-S2:
    --board adafruit_feather_esp32_s2  Adafruit Feather ESP32-S2
    --board esp32s2_bare               Espressif ESP32-S2 (bare)
    --board feather_s2                 Unexpected Maker FeatherS2 ESP32-S2
    --board msr207_v42                 MSR JM Brain S2-mini 207 v4.2
    --board msr207_v43                 MSR JM Brain S2-mini 207 v4.3
    --board msr48                      MSR JacdacIoT 48 v0.2
  ESP32-C3:
    --board adafruit_qt_py_c3          Adafruit QT Py ESP32-C3 WiFi
    --board esp32_c3fh4_rgb            ESP32-C3FH4-RGB
    --board esp32c3_bare               Espressif ESP32-C3 (bare)
    --board esp32c3_rust_devkit        Espressif ESP32-C3-RUST-DevKit
    --board esp32c3_supermini          ESP32-C3 SuperMini
    --board kittenbot_grapebit_esp32c3  KittenBot Grape:Bit ESP32-C3
    --board seeed_xiao_esp32c3         Seeed Studio XIAO ESP32C3
    --board seeed_xiao_esp32c3_msr218  Seeed Studio XIAO ESP32C3 with MSR218 base
  ESP32:
    --board esp32_bare                 Espressif ESP32 (bare)
    --board esp32_devkit_c             Espressif ESP32-DevKitC
  ESP32-S3:
    --board esp32s3_bare               Espressif ESP32-S3 (bare)
    --board esp32s3_devkit_m           Espressif ESP32-S3 DevKitM

fatal error: invalid board id: cardputer
```

I feel like this is inheriting the flash info from the arch here: https://github.com/microsoft/devicescript-esp32/blob/main/boards/esp32s3/arch.json#L16

Is it possible to overwrite this in the board config somehow? Am I on the right track? Is this possible?

reggi, 2024-01-31:
Even the Espressif ESP32-S3 board here has a pin 35 in the image so this isn't something unique to the M5Stamp, right?

https://github.com/microsoft/devicescript/blob/main/compiler/src/boards.json#L593C21-L593C129

https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/_images/ESP32-S3_DevKitM-1_pinlayout.jpg

[image]

I don't know what a "flash" pin is 😭, does M5Stamp need it's own Arch? Or is the `ESP32-S3` configured wrong?

reggi, 2024-01-31:
@mmoskal is it possible adding pins `35-37` in this commit was a mistake?

https://github.com/microsoft/devicescript-esp32/commit/686d949eeddb156cf091ff52c80ed930f635e7a9#diff-e62f0ba616b5f5106735cfc02f64dc3c686ff8a10a1442f77c52275801e7c4c8R16

reggi, 2024-01-31:
Perhaps this line from the docs clarifies something, this is wayy new to me 🙃🤯 https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/peripherals/gpio.html

> SPI0/1: GPIO26-32 are usually used for SPI flash and PSRAM and not recommended for other uses. When using Octal Flash or Octal PSRAM or both, GPIO33~37 are connected to SPIIO4 ~ SPIIO7 and SPIDQS. Therefore, on boards embedded with ESP32-S3R8 / ESP32-S3R8V chip, GPIO33~37 are also not recommended for other uses.

reggi, 2024-01-31:
Possibly related / similar? https://github.com/microsoft/devicescript-esp32/issues/13

reggi, 2024-01-31:
@pelikhan Hey! Thanks for your work on this project! Obviously I'm excited to get js running on this thing, I don't think many people have (most of the code out there is in c++ via arduino) and I'm really glad this project exists. Just would love to know if my guesses above as to what's going on is correct, if so does a new release need to be made (something out of my hands), or is it possible I can override the flash pins arch config from userland?

mmoskal, 2024-01-31:
@reggi thank you for digging info on `GPIO33~37` - I think the upcoming 2.16.2 release should fix it - it now labels the pins in question as `octal-flash` so should let you have the config you want!

closed 2024-01-31 by mmoskal

pelikhan, 2024-02-01:
Feel free to contribute back into this repo if it easier for you. Thanks for circling back!

pelikhan, 2024-02-01:
Using https://microsoft.github.io/jacdac-docs/tools/device-registration/ we can register that M5 board so that it gets a nice icon in the dashboard.

reggi, 2024-02-11:
@pelikhan this wants a PAT, and I really don't wanna do that 🙃, is there another way? 

[image]

pelikhan, 2024-02-11:
Feel free to ignore it and do a PR with the board.json and picture. Thanks!

commit 68c331f3 (2024-01-31): sync devicescript-esp32 updates; fixes #679