microsoft/devicescript #684: Cardputer Keyboard?

reggi, 2024-02-06:
https://docs.m5stack.com/en/core/Cardputer

[image]

[image]

74HC138 CMOS?

Seems to be a rust implementation here: https://github.com/syurazo/cardputer/blob/master/src/keyboard.rs

Pins as seen here: https://github.com/m5stack/M5Cardputer/blob/889cbd7b9d8c395205e22e924a383bd07089b14c/src/utility/Keyboard.h#L28C1-L29C62

```ts
const output_list = [8, 9, 11];
const input_list  = [13, 15, 3, 4, 5, 6, 7];
```

I've tried everything in the docs. I'm new to keyboards, serial, and i2c I have no clue to get button presses to register a console log. 

Is this something currently supportable by the project? I'm just worried that somewhere down the line devicescript will just not work for some piece of hardware on the cardputer. Can someone working on the team confirm that keyboard, speakers, and mic can all work? I got the LED on the board to work and the LCD so far.

pelikhan, 2024-02-06:
Let's see:
- you will need to add i2s support in a similar fashion to i2c to support the mic.
- you should be able to port the Rust driver to devicescript using the gpio module and detect the keystrokes.

> can act as an external keyboard over USB (is this what a HID Keyboard is)

Yep.

> Is it possible to do both via devicescript?

We do have hid keyboard driver in jacdac (https://github.com/microsoft/jacdac-c/blob/main/services/hidkeyboard.c) but i don't think it is fully wired up for devicescript. @mmoskal 

Anyhow, it's gonna require work and somewhat getting deep in building devicescript itself.

reggi, 2024-02-07:
@pelikhan thanks for this, I'm digging into the source. I'd down to contribute to the project. 

All I want right now is to be able to press any key on the device and `console.log`. This is not HID, just normal internal keyboard. 

I can't find this code: https://microsoft.github.io/devicescript/api/clients/keyboardclient

When I search the project for `KeyboardClient` nothing shows up 🙃

Where do I start?

pelikhan, 2024-02-07:
> All I want right now is to be able to press any key on the device and console.log. This is not HID, just normal internal keyboard.

For that you would need to implement the pin scanning code, just like the rust sample you mentioned.

reggi, 2024-02-11:
@pelikhan I got this working so far for this device I got the following working:

[x] Onboard LED
[x] LCD 16bit colors
[x] Keyboard

I'd like to put all this in a module that people can reuse. I think I'm interested in making something more custom. 

However modules like `@devicescript/drivers` aren't in the npm (or github) registry. 

How can I make my code into a package?

pelikhan, 2024-02-11:
Note that we'll be happy to take your drivers contributions back into @devicescript/drivers as well!

reggi, 2024-02-11:
Thanks done and done

https://www.npmjs.com/package/cardputer
https://github.com/reggi/cardputer

closed 2024-02-11 by reggi